atuline/WLED #162: Microphone not working

zx804262156, 2022-01-12:
SR-WLED beta version 0.13.0-b6 microphone does not work.
Microphone type: Generic I2S, Generic I2S With Mclk, Generic I2S PDM. I have tried all of these options to no avail.

The microphone I'm using is INMP441, which works fine for me in SR-WLED beta version 0.13.0-b3.

**WLED version**
 - Board: [default_envs = soundReactive_esp32dev]
 - Version [SR-WLED beta version 0.13.0-b6]
 - Format [ Binary and self-compiled]

Thank you for your help!

atuline, 2022-01-12:
Join up on the Discord to discuss. Looks like you compiled with the proper environment. Not sure if you performed a factory reset.

For binary, see https://github.com/atuline/WLED/wiki/Installing-and-Compiling

zx804262156, 2022-01-12:
I have performed a factory reset

rrlevy, 2022-01-16:
I'm also having mic issue with 0.13.0-b6

I'm using an analog mic (MAX9814) connected to ESP32 VP Pin (GPIO36).

I can see the waveform arriving at the ESP32 pin with my oscilloscope, but setting up Sound Input Settings won't work.
Even with 0 Squelch and 255 Gain, at Gravmeter effect there is NO MOVEMENT in the LEDs.

rrlevy, 2022-01-16:
I also just flashed it with 0.13.0-b4 and the mic works fine there

jonandel, 2022-01-16:
AGreed - I built one yesterday 0.13.0.b4 ESP32, SVP pin - and it works fine.  The gain was a bit low at first (default in WLED was 10?)

atuline, 2022-01-16:
> I'm also having mic issue with 0.13.0-b6
> 
> I'm using an analog mic (MAX9814) connected to ESP32 VP Pin (GPIO36).
> 

How exactly did you flash it?

rrlevy, 2022-01-16:
I downloaded the binaries from the Releases page.

I tried first flashing with ESPHOME and also later tried with ESPTOOL command line. Also flashed the v4 bootloader.

I also tried flashing the beta4 and later upgrading OTA with beta6, and the mic stops working after the update.

rrlevy, 2022-01-16:
Here is the serial plotter of the same board:

[image]

The mic is working on the board. There seems to be something with the code.

rrlevy, 2022-01-16:
@atuline 
I don't know what happened but just after doing further tests, the mic simply started working on beta6....

I had tested everything before and I'm not sure what was the reason for the change. I had just reflashed the binary after running the serial plotter.

Just to make sure it really worked, I erased the flash memory, reflashed bootloader (v4) and beta6 binary, and it all worked again.

Sorry about that.

FHeilmann, 2022-01-21:
One thing that's particular with the updated microphone code is that any change to the configuration requires a hard reset (via reset button) or powercycle. The reason for this is that a software reset (pressing reboot in the UI) doesn't properly reset the I2S peripheral of the ESP32 and leaves it in a disfunctional state. This is unfortunately a quirk of the underlying framework and something we cannot fix, if you have any suggestions how to clarify this, I'll be happy to hear them.

Bests,

Florian (Haribro)

dosipod, 2022-01-22:
Hi guys , I just moved two ESP32 that are working fine with music reactive features   via line-in  from  sr 0.13.0-b4 to  sr 0.13.0-b6 and no more leds reaction . I moved back to  0.13.0-b4 and the issue is fixed . I have not changed anything after each update

atuline, 2022-01-22:
When you flashed, did you:

https://github.com/atuline/WLED/wiki/Installing-and-Compiling#flashing-esp32-binaries

dosipod, 2022-01-22:
I only flashed the binary with OTA , as it worked on its own the 3rd time  I assumed it was just as you said a matter of time  from when flashing is finished , I will redo the same with esptool because the guys reported issues with ESPHome Flasher . Is that what you want me to do ? I also have 3 other wled setups that I can test beside the two I a have already done testing

rrlevy, 2022-01-22:
> I only flashed the binary with OTA , as it worked on its own the 3rd time I assumed it was just as someone said a matter of time from when flashing is finished , I will redo the same with esptool because the guys reported issues with ESPHome Flasher . Is that what you want me to do ? I also have 3 other wled setups that I can test beside the two I a have already done testing

@dosipod I had the exact same issue. Had to reflash by erasing, flashing bootloader and flashing binary and it finally worked.

dosipod, 2022-01-22:
I think it is easier then I expected , just pressed reset after the OTA update worked ( reset by pressing the EN with no power recycle  , I think that should be mentioned to the guys as a first step to avoid complication

victorcampeanu, 2022-01-25:
0.13.0-b4 works,  0.13.0-b6 works after OTA update then doesn't after a power off. restoring to  0.13.0-b4 works again

atuline, 2022-01-25:
Please see the 'Its not working page', and the "Installing and Compiling" page.

victorcampeanu, 2022-02-08:
it starts to work after a long press on reset but that is untill the next power off, you have to long press EN again.
also there is alot of noise, with room quiet it lights up randomly. 
restoring 0.13.0-b4 fixes all isuues

btittelbach, 2022-04-22:
don't have a lot to contribute but also observed that upgrading to 13.1 on esp32 cololight-pro analog microphone stopped working where previously it worked fine. Downgrading to 12.0 makes microphone work again.

smith844, 2022-05-22:
I concur, the microphone stops working under 0.13.0-b6 but works as advertised with b4.  I also have generic analogue set with MAX4466.  I have reverted to b4 and it works fine.  looking forward to the undoubted fix. 
Thanks to all for the fine work.

softhack007, 2022-06-28:
We recently found out that analog microphone input sometimes does not work when you also have "analog buttons" (aka potentiometer) configured.
I2S uses the ADC1 unit; when there is another ADC1 operation by analogRead() then two drivers seem to be in conflict, as both want to use ADC1 but with different settings -> as result, I2S delivers silence.

Do you have "analog buttons" configured in your LED preferences? If yes - please check if the microphone works after changing all buttons to "disabled" and reboot via reset button.

closed 2022-07-21 by ewoudwijma