atuline/WLED #166: ESP32 Brownout detection

ZorK766, 2022-01-29:
Hi, I made a first project box with ESP8266 D1 mini for controlling WS2812 3 meters 60 Leds/m using WLED. Works great, but I made another "improved" one, including a microphone module for sound reactive effect (Fork) and a Relay for power saving.

 This new box works as intended when tested with 30 Led, powering via Micro USB Port. But when I plugged a 5V 10A alimentation, it refuses to work. I figure out why: for using Sound reactive effect, I switch on an ESP32 Vroom 32 in D1 form factor from AZ-Delivery. And this module seems to not operate when input tension is superior to 5.2 V.

 I Tried with two different 5V 10A power supply (One from  LEICKE and the other from BTF-Lightening). Both outputs around 5.37V without load. On my second box, I try with an adjustable power supply, if I power it above 5.2V the ESP32 won't boot. But if I replace it in the same box with the ESP8266 (claim to accept 6V) it boots correctly.

 QUESTION: do you have tips or idea for me to solve my problem ? I thinked to use a voltage regulator like A7805, but cause my PCB is already soldered, that'is not really simple to modify it in such small compact form factor... Some peoples suggere me that ESP32 have brownout detection (ESP8266 hasn't). Is ther a way to modify this value in sources, then re compile it ? Regards

[image]
.

ArduinoNerdBheeshma, 2022-01-29:
I had a similar problem where my esp32 won't boot properly when using external power (5v 20A). A simple fix I found was to add a capacitor at the esp's Vin and Gnd pin (I used a 1800uf 16v cap but feel free to experiment) . Note that capacitor should be at the Chip's side and not the power supply side.

atuline, 2022-01-29:
Join up on the Sound Reactive Discord channel for support. We typically use github for software issues, i.e. patches or confirmed bugs.

johnsonej23, 2022-01-29:
Hey Zork, where did you get that case and breakout board?

ZorK766, 2022-01-29:
@ArduinoNerdBheeshma : Already have a capacitor on my board, the problem wasn't a "dirty" power signal but an "overvolatage" problem. Easily verified when powered by an adjustable power supply. It's working at 5,2V and stop working as soon as I go over 5,2V by adjusting the power supply knob.

@atuline : Thanks, sorry about that. I close the issue as it's solved.

@johnsonej23 : Cases found on Amazon. Search for "Project Box". For the breakboard, I bought a pack of different sizes on AliExpress. Not really the most practical to use, but a good way to "validate" à prototype. I made first testes on a traditional breakboard, then with this kind of "PCB". Thanks to that I figure out some issues like this one, an't validate some components or connectors placements. Now I will test it in place like that, and then create a real PCB on EasyEDA and order some of them by JLCPcb, with probably some CMS component on it to reduce the size even more. Finishing by printing a custom 3D case and i think It will be great when after that I'll feel satisfied ^^

SOLUTION :
I solved my problem inserting a 1N4001 Diode between 5V power and ESP32 VCC Pin. Now the ESP32 is happier with 4,7 V when powered by my 10A Alim, and the Led Strip still got full power. Downside is with this setup, Led strip isn't powered anymore when I powered the box via ESP's micro USB. But not really a big concern, cause now my setup is ready to install.

Regards.

closed 2022-01-29 by ZorK766

ArduinoNerdBheeshma, 2022-01-30:
Some metal power supplies have a voltage adjustment knob. You must be using a power brick then

ZorK766, 2022-01-31:
@ArduinoNerdBheeshma Yes, I've got one of this 5V 6A meanwell type of metal block with screws terminal. It supposed to go in my 3D printer for powering an integrated Raspberry Pi, his screen and some other 5V add-ons. I want to keep it for that purpose, cause it makes one year I'm supposed to finish the customization of this printer. 
I thought to use it like you were suggesting it, but after testing yesterday, it seems that one or two amps are really enough for what I need: 
This Wled Box was intended to be installed behind my bedroom TV. There are 168 WS2812 Leds, in an custom aluminum frame. 
Thanks to the Wled current limitation settings, I figure that I will never go above 1 or 2 Amp because It is enough to light all my wall perfectly and more is really an eye pain situation. We watched TV like this during two hours yesterday and I felt perfect with 1000mA current limitation in Wled.
So in fine, I will probably just use a simple 2A phone charger. And it t will be really simpler to install behind the TV. ^^
I could even power all that by one of the USB of my TV. But I Will be quieter with a two amp adapter feeding the box. Like this, I would be able to use the lights without turning out the TV. One of the main goal off all this stuff is to use Sunrise and Sunset lights to accommodate my sleeping and waking times ^^
Here are images of my frame and on image of the device running in half brightness setting in Wled and 1000mA current limit set in config. In reality, the result is really more pleasant that on this picture, but it would give you the idea... 

[image]

[image]
.