atuline/WLED #167: Soundreactive wled noise issue

ArduinoNerdBheeshma, 2022-01-29:
**Describe the bug**

Hello I'm making a diy nanoleaf with about 565 neopixels, a 5v 20a power supply, esp32 dev and a max 4466 electret mic. The esp boot cycles on power up but sometimes turns on fine. But when I turn on a music effect the leds bounce around in complete silence. Decreasing the gain on the max4466 does solve this issue but it becomes too un sensitive. Pls help
**To Reproduce**
Squinch:19 gain:5 ADC:34 
Taped to my TV's speaker
Depth: halfway
Gain on mic board: unknown

**Expected behavior**
The whole strip is off at silence’’
Reacts to music that is a bit faint too*
Reacts to bass 
Reacts to bass in rapid succession
Maybe reacting to lyrics as well*
Maybe react to speech in the bg*
* = optional but will be cool if possible
’’ = if possible but needed a bit more than the *'s

**WLED version**
Esp32 dev
 - Version   latest
 - Format   web browser flash
**Additional context**
I'd maybe also use line in but haven't tested it out yet. If I opt using line in, how do I play music on speaker as well as headphone jack?

ArduinoNerdBheeshma, 2022-01-30:
Update- Line in works but with bass only. Leds react only when bass is played. Also I misspelled depth it's intensity

atuline, 2022-01-30:
Some questions:

- Can you also test with a short strip, let's say 30 leds. I don't have 565 laying around.
- How about a specific animation? I can't attempt to re-create if we don't have an apples to apples comparison.
- Run the test sketch as linked below.

You might want to check the quality of your power, because a squelch (and not squinch) of 19 should cut out a considerable amount of background noise. The microphones do not work below 60Hz.

Please note that there is a considerable amount of interference from the WiFi on the ADC. See [Noise Article](https://electronics.stackexchange.com/questions/368867/esp32-microcontroller-generates-noise-on-microphone) and that a better bet is to use an I2S microphone such as the INMP441 or ICS-43434. The noise issues are documented in our SR Wiki.

As for reacting to music that's faint, you would use Automatic Gain Control, however that's a tricky beast, because the device will no longer differentiate between quiet passages in a song and louder ones. Everything will look loud.

I recommend that you use a test sketch such as [this sample sketch ](https://github.com/atuline/Arduino-Sampling/blob/master/analog_sample/analog_sample.ino to ensure you don't have issues.

Please compare your results to [Noise and Spikes article](https://github.com/atuline/WLED/blob/assets/docs/Noise%20and%20Spikes.pdf).

dosipod, 2022-02-01:
Hello ,

I think I have  seen you asking this question in wled  discord  or forum , please drop by the soundreactve  channel as I have already replicated your setup with  max4466 , 512 pixel  ws2812b  matrix and  line-in on multiple MCUs  with very good results and no noticeable ambient noise or interference issues  . 

As this seems to be a user setup/hw related , I think  having a  github bug report for WLED it is not accurate  , kindly  close the same . 

If you drop by wled discord or forum and would like me to directly assist in isolating your issue then I am available with the name **ALDIY**  . 

Cheers ,
AL

softhack007, 2022-03-03:
Apart from that, 
there seem to be some sources of "artificial spikes" in the code for reading the MIC input and performing FFT. I managed to eliminated some of them in my own copy of the code, however it still needs a lot of cleanup before I can make a PR. 

Some problems I found, especially for the FFT analysis:
- substracting the average "DC component" from the input - this should be done only once, as the last step of the processing chain. This operation is basically an **IIR filter**, and it changes the spectral composition of the signal. The ArduinoFFT lib has a function `FFT.DCRemoval()` that does a much better job.
- be careful with `abs()`. It looks innocent, however when applied to a signal that also has negative values, it flips your signal over to the positive half whenever there is a zero crossing --> adds noise to the FFT.
- finally, repeated `i2s_adc_disable()` / `i2s_adc_enable()` seems to add pop/crack sounds when using an analogue MIC. The reason might be that the ESP32 i2S driver is not aware of the DC offset (Vcc/2) in the MIC input. After `i2s_adc_disable()` the input is set to 0, however when switching it back on, the reading jumps back to the DC offset  of Vcc/2 -> effectively a DIRAC impulse that burns itself through all frequency bins in an FFT.

I'm trying to make a PR with my proposal on how the code might be improved. However it may take some time...

atuline, 2022-03-03:
Sounds like some real good additions/fixes.  You might also want to talk to Haribo on the WLED Discord. He's been involved with the sound stuff as well.

softhack007, 2022-03-03:
Thanks, will try to chat with him 👍 once I have cleaned up my patched code a bit.

Btw, I've spotted something strange in the `agcAvg() `code:
https://github.com/atuline/WLED/blob/575853db3af7ef4402b3f129c1a6012a23a7ed4e/wled00/audio_reactive.h#L203

Maybe this should be ` if (tmpAgc > 255) tmpAgc = 255; `?
At least it looks strange to me, that the AGC multiplier is reset to zero each time that it reaches 255.

atuline, 2022-03-03:
That's most likely my fault, and I'll have to try and remember what I had in mind there. I hastily added the tmpAgc variable once I realized the multi tasking nature for use with segments and that I could not have interim values of sampleAgc.

softhack007, 2022-03-16:
Back to the question from @ArduinoNerdBheeshma - 
Another problem with analogue microphones, especially MAX4466,  is they are really sensitive to Vcc voltage "noise" ripples. 
I was able to reduce the wifi-related noise with a capacitor. However also make sure _not_ to power the LED strip directly  from the ESP32 board (neither from 3.3V nor 5V).  Each time that the strip lights up, it draws power and this will add **loud** noise to the microphone ADC readings.

* use an external power supply for the strip (but remember to connect grounds)
*even better, to avoid all the mic troubles - invest a few bucks into an I2S digital mic, like the INMP441. It's a **huge** improvement; no more "LEDs jumping around in complete silence".

commit 24e3674a (2022-03-06): Merge pull request #178 from softhack007/analogue_mic_processing_improvements

Analogue mic processing improvements (fixes #167)
closed 2022-04-07 by THATDONFC