atuline/WLED #168: Alex can't find the ESP32 NodeMCU Soundreactive v.0.13.0-b6

MasterDevil1980, 2022-01-29:
Hello everyone,
I have an ESP32 NodeMCU and am running it with the WLED Soundreactive version v.0.12 and also the current v.0.13. The option to emulate for Alexa is active but my Alexa is not finding any devices. I previously had the ESP8266 running with Wled and Alexa recognized the device directly.
What am I doing wrong.
I hope someone has a solution.

Thank you for your help!

atuline, 2022-01-29:
We've disabled several services by default with SR WLED. You may need to enable it and then compile.

MasterDevil1980, 2022-01-29:
I turned on the option and gave it a name. But Alexa can't find the device. Everything worked with my Esp8266. Do I have to activate anything else in this version?

MasterDevil1980, 2022-01-29:
you mean i have to edit the file alexa.spp or what?

atuline, 2022-01-29:
For support type questions, please join the Discord, as there's a community of folks on there. In the meantime, see:

https://github.com/atuline/WLED/blob/ce28964e0462ee76451142b2b82891daecf2be16/wled00/wled.h#L30

and we don't support those services.

closed 2022-01-29 by MasterDevil1980