atuline/WLED #169: On some Effects the last led does not work. And connection problem.

LEDNoob71, 2022-02-05:
I am using the Esp32 dev.
I have 31 LEDs on my strip.

On some sound reactive effects the last LED does not work.
I tried different effect settings but the last led does not light up.
These are the effects:
gravcentric
gravcenter
midnoise
gravfreq
noisemove

is this a bug?

The second problem is:
sometimes if i try different effects and settings, the wlan connection to the strip is lost. On my router i see that wled it always connected. But the App shows "connection lost" and on the Windows PC no connection possible to the IP. In this case i must reset the ESP32. 

How can i solve this problem.

atuline, 2022-02-05:
The effects that emanate from the center will require an even number of leds.

As for the connectivity, we've inherited this functionality from AC's master. I've heard that Disable WiFi sleep mode helps.

LEDNoob71, 2022-02-05:
Thank you for your prompt reply.
And thank you for your excellent work on the project.

LEDNoob71, 2022-02-07:
Disable Wifi sleep does not help.
I made my settings in wled and all works fine. after few minutes i can not connect to wled. App and Windows PC!
In my Router config i see wled is in the network, but no connection is possible.
i am using the MH-ET LIVE MiniKit ESP32 Module.

What could be causing the problem?

atuline, 2022-02-08:
We've also found more problems with AP mode. I use a static IP address and if I wanted to keep it live for a while, I'd check that Disable Sleep mode box.

LEDNoob71, 2022-02-08:
After few tests i know that the esp32 module is the problem :(
I have flashed a simple web server program from the arduino examples. the esp32 still loses the connection after a random period of time. sometimes in few minutes and sometimes a little longer. But the esp32 ip is still visible on my router as connected. 
No Pins are connected. Power is supplied via the USB port from the Windows PC.

I know, this is not longer an issue from wled. But does anyone have an idea why this error occurs?

softhack007, 2022-05-23:
about the first issue - noisemove was having an off-by-one bug, fixed in dev https://github.com/atuline/WLED/commit/4454d4bfaee59cb342128b37b7e096eca04d7fff.
For the other "centric" effects, an even number of LEDs is assumed - otherwise you cannot "mirror" the effect at the middle of the strip.

softhack007, 2022-07-14:
Closing, as the author confirmed this is not a SR WLED specific problem.

closed 2022-07-14 by softhack007