atuline/WLED #170: Auto gaint

ddijanic, 2022-02-07:
auto gain. You were great to have, software auto gain. I've seen with these devices have,
https://a.aliexpress.com/_mK5WoM4
. the question is whether it is difficult to add that ...

atuline, 2022-02-08:
AGC is something I've struggled with for some time now and I have routines that do that, but the leds will be 'loud' during quiet passages in an otherwise loud song, as much as they'll be 'loud' during both loud and quiet songs. In additon, the ability to window each frequncy bin (with, let's say an exponential filter) and with a reasonably accurate representation of energy levels across those frequencies. . . let's say using some form of AGC. . . well . . . that's been a real challenge for my high school math abilities. 

Am always up for a good PR though.

softhack007, 2022-04-11:
Experimental feature available in https://github.com/atuline/WLED/tree/dev - currently supported by only a few effects.
PR https://github.com/atuline/WLED/pull/195
PR https://github.com/atuline/WLED/pull/196

netmindz, 2022-04-14:
To solve the LEDs being "loud" I think we just need to adjust the gain over much longer time period, e.g over the last 5 mins.

softhack007, 2022-04-14:
> I think the answer would be Proportional Integral control

If I understood Wikipedia correctly, my new implementation is basically a "P controller", just with a very complicated way of choosing the proportional gain (Kp). I will check over the WE if adding an "integrational" part - which makes it a real PI - can improve stability. 

>  I guess I should've taken the Controls option back in the day at university.

Same here 😉 - in the good old days at university, I chose "audio processing" instead of "control theory" because someone told me that 'the math might kill you'

> just need to adjust the gain over much longer time period, e.g  over last 5 mins

This can be done independly of the AGC code, simply by slowing down the frequency at which AGC runs. Something like` if(millis() - last_time < SlowDown) return();` should do the trick.

I'll keep you informed about my PI experiments.

atuline, 2022-04-14:
In the meantime, here's some of the work I did on that:

https://github.com/atuline/FastLED-SoundReactive/blob/master/agcPI/agcPI.ino

I think I've got a better version kicking around on my hard drive as well.

Also, Ben Hencke of Pixelblaze made good use of PI Controls with his sound reactive code. Guy's a genius.

netmindz, 2022-04-15:
I have been very impressed with audio responsive stuff on pixelblaze to be fair

softhack007, 2022-04-16:
Please try my latest PR https://github.com/atuline/WLED/pull/200 (should be merged into dev soon), and tell me what you think. I managed to stabilize the gain factor, while still being able to follow volume changes within a few seconds.

softhack007, 2022-05-06:
Available in development (`dev`) branch https://github.com/atuline/WLED/tree/dev

closed 2022-05-06 by softhack007