atuline/WLED #172: Edges of 2D effects using blur bleed onto opposite edge of matrix

ceeeekay, 2022-02-13:
**Describe the bug**
Edges of any 2D effect with a blur component will appear on the opposite side of the matrix when blur is set to moderate/high values and the effect is drawn at the panel margin. Is there any way to stop the blur from wrapping like this?

E.g., 2D DNA:
[image]

Note: Ambient lighting was quite high for this pic. The artefacts are a lot more pronounced in person, and as they're in motion they're extremely noticeable.

As far as I can tell this occurs on all 2D effects, including SR, but the 2D DNA effect was a lot easier to show the problem.

**To Reproduce**
Run the following preset. 

    {"on":true,"bri":251,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":256,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"cct":127,"col":[[255,160,0],[0,0,0],[0,0,0]],"fx":152,"sx":128,"ix":241,"c1x":255,"c2x":123,"c3x":120,"pal":22,"sel":true,"rev":false,"rev2D":false,"mi":false,"rot2D":false}]}

**Expected behavior**
Effects should display cleanly with no bleed/artefacts on the opposite side

**WLED version**
 - Board: ESP32 dev + external INMP441
 - Version: 0.13.0-b6 2112080
 - Format: Binary from releases page

Thanks!

dosipod, 2022-02-22:
Hi , I tried the preset you posted with sr13.0.b6 on 3 different ws2812b eco matrices ( 16x16,  8x32 & 16x32  ) and I have not noticed anything strange with the matrix edges. I have also tested multiple random effects with no issues that I could see , I had a friend look at it but it was so bright that we could not really tell if there is an issue or not . 

It could be my bad eyes ,please check the pics if you see an issue . The preset you sent is on high brightness and mostly white which could be an issue for certain configuration/hw setups . 

Also just a note if you are also testing this on ws2812b eco matrix to keep in mind that they are not really reliable as  I have one that is behaving strangely after only 2 months of continuous use .( that is the reason for the reduced cost but also for the shorter warranty period )  Do you have more then one matrix that you can test and verify preferable on a different MCU ?  

In all cases good luck and if you are still facing this issue then please drop by the wled discord or forum to  hopefully solve it  , 
[image]
[image]

ceeeekay, 2022-02-22:
@dosipod I'll follow this up on discord, but in the mean time I've tested the same preset with a couple of 8x32 panels in a 2x1 array, and get the same issue.

**What's strange is if I tick the rotation box in the segment interface, the artefacts disappear. Untick it again so the rotation is normal and they reappear.**

Vid below, where you can see the artefacts up the right side. First normal, then rotated, then back to normal.  All I'm doing  is ticking the rotation box.  I really don't think it's hardware-related.

https://i.imgur.com/yHL4cXU.mp4

dosipod, 2022-02-22:
Yes I can now clearly see the issue in the video .

ceeeekay, 2022-02-22:
@dosipod I've joined the WLED Discord but I don't see a channel when clicking your link - just a Discord error page saying "no text channels". Do I need an invite to another channel first?

softhack007, 2022-09-07:
closed by https://github.com/atuline/WLED/pull/224.

closed 2022-09-07 by softhack007