atuline/WLED #174: VBAN Receiver (VB-Audio)

reptil1990, 2022-02-22:
Hey Guys,

I had an Idea what could be quite usefull for a lot of users.

"VoiceMeter" is a digital mixer for windows and with that you can manage a lot of audio divices and mix it together.
This program is also having a protocol called "VBAN" It is a form of UDP audiostream.

Now it would be really cool if the WLED controller could act as a reciever and get the soundinforamtion directly from this stream.
I already found a project that is acting as an reciever to build speaker: https://github.com/flyingKenny/VBAN-Receptor-ESP8266-I2S

We just would need a conversion from this UDP audio stream into WLED.

No external Hardware would be needed.

The documentation to the Protocol and some examples can be found here: https://vb-audio.com/Services/support.htm

Thank you for the awsome work.

Greetings reptil1990

jonandel, 2022-02-23:
If you've got Windows, could you not send VM directly into LEDfX....which can control WLED ?

reptil1990, 2022-02-23:
Yes that's true but it's another program running! And since the digital mixer is running anyway it would save recources.
VBAN also have a Linux implementation and then it can also work under linux

netmindz, 2022-02-24:
This might already be supported. Try using this project you listed and then connect to the ESP32 as if it were a digital mic. There was discussion recently about using a pair of controllers to do Bluetooth to WLED

reptil1990, 2022-02-24:
I would be more than happy to test anything you want to have tested! If it is not to much effort to implement a proof of concept I can test it, I have all the Hardware and Software ready to go.

atuline, 2022-02-24:
We just need someone to create a VBAN branch, and do a PR with that functionality added. Then . . . the testing.

netmindz, 2022-03-04:
I would strongly advise the dual microcontroller approach rather than trying to stream uncompressed audio, FFT and LED output all on single esp32

atuline, 2022-03-04:
Yea, we gave that some thought, and uncompressed audio is just not going to happen.

closed 2022-07-14 by softhack007