atuline/WLED #175: E1.31 Broken in fork

netmindz, 2022-02-23:
**Describe the bug**
When using E1.31 the address to pixel is broken, starts off correct, then some don't map, multiple pixels on the same address etc

**To Reproduce**
Light up one pixel at a time using e1.31 on the sound responsive branch with 2D matrix, then try the same with non-SR

**Expected behavior**
One dmx address should only ever match one pixel

**WLED version**
 - Board:  ESP32
 - Version 0.13.0-b6
 - Format Binary

mase76, 2022-02-24:
Is this similar to this? https://github.com/smeighan/xLights/issues/2832
I use DDP, but this should not make a difference.

JvPeek, 2022-03-07:
Same here. pixel 1 is correct, pixels 2-6 are broken, 7 and 8 are working again. I used the same jinx mapping i used for the regular WLED version. After giving it a try with WLED b7 i switched to sound reactive b6. Right out of the box the mapping was still correct. it broke when i set up my LED matrix which consists of multiple 32x8 panels.

JvPeek, 2022-04-23:
I made a short video detailing the error.
https://youtu.be/OjjYAwPSKbA

Angel-1024, 2022-09-11:
Yes same to my matrix. But this mapping bug appear only if 2D matrix selected in settings. If i set 1D strip in settings (but physically leave 2D matrix connected  to ESP32) remote control with E1.31 will working fine and mapping become correct, but in this case we lost local 2D visualisations.

netmindz, 2022-09-11:
> Yes same to my matrix. But this mapping bug appear only if 2D matrix selected in settings. If i set 1D strip in settings (but physically leave 2D matrix connected to ESP32) remote control via E1.31 will working fine and mapping become correct, but in this case we lost local 2D visualisations.

That's useful to know, guess can be workaround while we find the bug and also gives a clue as to where it is

Angel-1024, 2022-09-12:
> That's useful to know, guess can be workaround while we find the bug and also gives a clue as to where it is

today i tried different modes on my matrix. 
My physical matrix Width 15 and Height 20 pix ws2812, connected columnwise top left is 1st LED. Map same in Jinx!  
Wled set in LED settings tab: fact. results internal Wled FX, remote control E1.31 via network or tpm2 player from sd card, attached to ESP32 (https://github.com/Aircoookie/WLED/pull/2413#issuecomment-1236206323) 
Results, hope it helps to fix it.

WLED settings 1D 300 LED: WledFX (mean internal 2D effects only) wrong map; E1.31 (2D Jinx) OK vert orient.
WLED settings 2D W15xH20,horizontal: WledFX wrong map, E1.31 OK vertical orientation
WLED settings 2D W15xH20,horizontal,transposed: WledFX OK vert orient, E1.31 wrong map
WLED settings 2D W20xH15,horizontal:_____________WledFX OK horizontal orient, E1.31 OK vert orient
WLED settings 2D W20xH15,horizontal transposed: both WledFX and E1.31 wrong map
WLED settings 2D W15xH20,vertical:________________WledFX OK vertical, E1.31 wrong map
WLED settings 2D W15xH20,vertical transposed:     WledFX wrong map, E1.31 OK vertical
WLED settings 2D W20xH15,vertical:________________both WledFX and E1.31 wrong map
WLED settings 2D W20xH15,vertical transposed:     WledFX OK horiz orient, E1.31 OK vertical

giraff11, 2022-09-21:
Also having issues with E1.31 (and DDP) on 0.13.2
I have a single 63 LED WS8212B strip connected, everything works fine on WLED SR.
However using Hyperion via DDP or LEDFx via E1.31, WLED only responds to the last half of the commanded LED colors and mirrors it over the first half... that is, LEDs 32-63 are mirrored over LEDs 1-31

tbuaas, 2022-09-27:
For what it's worth, I built the latest code in the dev branch the other day, and this 2d matrix issue when controlling via DDP or E1.31 seems to be fixed. I too experienced this issue in the latest release build (0.13.2-SR).

softhack007, 2022-10-13:
@netmindz please check if the problem still exists in out `dev` branch.

Angel-1024, 2022-10-15:
WLED 0.13.2.1-dev (build 2209201) this problem has fixed. But whenever i using LedFX animation is rotated on 90 degree, on PC screen animation displayed in horizontal orientation but on led matrix displayed in vertical orientation. And bult-in "rain" effect falls from left to right, but "2D matrix" falls normal - from up falling down.
My matrix settings: width 15 height 20 leds, 
First led position: top right, 
orientation: vertical. Same as phisycal orientation.

softhack007, 2022-11-28:
Closing, as the original problem was solved with SR release 0.13.3. Please open a new issue report in case that you still experience the LedFx orientation problem.

closed 2022-11-28 by softhack007