atuline/WLED #184: INMP441 Not working?

Mazdaracer121, 2022-03-15:
My INMP441 not working.. I double checked my wiring, tried another Esp32…
How can I get it work?

wled-install, 2022-03-15:
INMP441 works definately very well

softhack007, 2022-03-16:
The INMP441 is definitely working with SR-WLED, and it's a great mic. Maybe you can find some clues in our Wiki:
- https://github.com/atuline/WLED/wiki/Digital-Microphone-Hookup
- https://github.com/atuline/WLED/wiki/Sound-Settings#how-to
- https://github.com/atuline/WLED/wiki/My--device--is-not-working

Please also check your WLED settings-> sound setting. They should match with the wiring of your mic. My settings look like this:
[image]

al3x1337, 2022-03-21:
Would like to add that my INMP441 that I ordered also doesn't work.
Wired 32/15/14 to the corresponding D14/15/32 pins on my ESP32S. Tried both 13-b4 and b6. Confirmed my settings and made sure in LED Settings those pins we not already assigned. My L/R and GND are grounded. 

I have a wire attached directly to the pads on the INMP441 and confirmed that there's continuity. 

Maybe I got a defective microphone?

softhack007, 2022-03-21:
🤔 hmmm ... ESP32**S** ... can you tell exactly with ESP32-S**n** model you have? like ESP32-S, ESP32-S2, or ESP32-S3?
https://www.espressif.com/en/products/socs

Depending on the model, a newer espressif32 platform (and ESP-IDF which is included) might be needed to support your hardware. For example, support for ESP32-S2 was added in ESP-IDF 4.2 - not sure what is the corresponding  `platform = `  for platformio.ini ...

* Didi you also try to compile the latest sources from the _dev_ branch?
* if you click on "Info" in the WLED UI, what do you see as "Environment" (last line) ?

omski, 2022-03-21:
> here is the Amazon link

it think you forgot the link

al3x1337, 2022-03-21:
Thanks for replying back with your settings and wiring pictures. As far as i can tell I think it's wired correctly:

VDD -> 3.3v
L/R -> GND
GND - GND
SD -> 32
SCK -> 14
WS -> 15

I will try installing b2 to see my results but I already think my microphone is defective and have another one on-route to confirm tomorrow.

al3x1337, 2022-03-24:
Just received another INMP441 and no luck still. On SR 13-b6 and a sound sound effect selected the LEDs randomly flash when there is no sound.

atuline, 2022-03-28:
Am now testing this branch:  https://github.com/atuline/WLED/tree/dev-v0.13.1-merge

INMP441 is working fine with settings from [al3x1337](https://github.com/al3x1337) on a LOLIN D32. I made sure not to get any flux in the microphone hole.

softhack007, 2022-04-01:
Two more thoughts on the wiring https://github.com/atuline/WLED/issues/184#issuecomment-1073945104

1. does the matrix really work for you this way? 
* No level shifter - risky not to have one, but it might work)
* missing a  buffering capacitor - please add one between 5v Vin and GND, especially when you "feed" the ESP32 from the same source. Typically poeople use a 1000 microFarad one. In case your power supply is not "smoothed", even 2200microFarad could be needed.
* missing  resistor on the data line (green) going from ESP to the matrix - Usually a resistor (300 to 500 ohms) is hightly recommended, for signal stability but also to ensure that the matrix does noch "suck" too much power from the esp32 pin.

2. Just for a quick shot, you may try to conned the "L/R"  mic pin to 3.3v instead of GND, which switches the input signal to the other channel.

. 

3. More a developer question (hi @atuline,  @FHeilmann, and others) - could it be possible that the NeoPixelBus wrapper grabs I2S0, so that we are later unable to connect to the microphone? I'm thinking about these two driver modes:

https://github.com/atuline/WLED/blob/2d4d6a2fc402cd7c88de30fa3d117fb392cc3fe8/wled00/bus_wrapper.h#L101

https://github.com/atuline/WLED/blob/2d4d6a2fc402cd7c88de30fa3d117fb392cc3fe8/wled00/bus_wrapper.h#L107

softhack007, 2022-06-26:
From SW point of view, the IMNP441 digital mic is well-rested, and works for many people. 

I can imagine two kinds of problems:
a) your mic is not able to use left channel (hw error), or other I2S communication problem
b) wiring issues, like using pins that are already in use for other functions (for example ESP32-CAM, where many pins are connected to the camera interface)

Please join Discord, where we have a "support" channel and the WLED community that could help with any kind of problem. https://discord.gg/4CQRmfR

Closing this issue as there seems to be no specific software problem. Please use discord to get support from the community.

closed 2022-06-26 by softhack007