atuline/WLED #187: Select preset with sound threshold

hobbypin, 2022-03-24:
When sound(volume) reaches a set threshold change to preset x. Once sound falls under preset for some delayed period of time return to previous preset. Very much like a button but triggered on sound level.

My use case is using SRWLED in a jukebox. nothing playing I use a simple attract mode. When music starts I would like to use one of the Sound Reactive effects. When music stops return to attract mode.

Wiring a button is possible but sound level trigger would make it a lot easier

pnichols77, 2022-04-03:
> When sound(volume) reaches a set threshold change to preset x. Once sound falls under preset for some delayed period of time return to previous preset. Very much like a button but triggered on sound [...]

Providing the media player youre using is visible in Home Assistant, you could use that  to auomate sound reactive effects when media is playing.

softhack007, 2022-08-12:
I think this is not possible currently. And it would not work with AGC enabled, which will become the new standard.

We take ~50 samples per second from the microphone, and AGC tries to balance input amplification so that samples use 90% of the availeable range which is 0...255 . Chances are very high that every possible value is used within a few seconds. 

What could be done in future:

1) once we have a working "noise gate" - raise a trigger when there is silence for some seconds

2) AGC uses an internal gain factor. This factor is highly depending on sound input characteristics, like microphone sensitivity and dynamic range of I2S readings. Currently we don't make the AGC gain value visible to users. We could - in principle - check AGC gain agains a user-defined threshold, and then raise some trigger when gain is _below_ for some user-defined time. "Below" as high gain indicates a very low input signal so low gain means there is stronger sound.

actually there are no plans right now to create something like this, but it could be possible from technical perspective.