atuline/WLED #188: Seven Segment Display Reloaded Usermod

al3x1337, 2022-03-28:
**Is your feature request related to a problem? Please describe.**
Seven Segment Display Reloaded from Aircoookie/WLED:master is missing from SR

**Describe the solution you'd like**
Seven Segment Display Reloaded added to this repo

**Describe alternatives you've considered**
I've tried to compile from scratch when I noticed the usermod is missing from this repo. I've added the usermod folder and made identical changes in const.h and usermods_list.cpp to those in the main WLED repo but get the follow error when compiling:

```wled00/../usermods/seven_segment_display_reloaded/usermod_seven_segment_reloaded.h:394:27: error: 'stateUpdated' was not declared in this scope
             stateUpdated(1);
                           ^
wled00/../usermods/seven_segment_display_reloaded/usermod_seven_segment_reloaded.h: In member function 'virtual void UsermodSSDR::onMqttConnect(bool)':
wled00/../usermods/seven_segment_display_reloaded/usermod_seven_segment_reloaded.h:463:75: warning: format '%S' expects argument of type 'wchar_t*', but argument 4 has type 'const char*' [-Wformat=]
       sprintf_P(subBuffer, PSTR("%s/%S/+/set"), mqttDeviceTopic, _str_name);
                                                                           ^
wled00/../usermods/seven_segment_display_reloaded/usermod_seven_segment_reloaded.h:463:75: warning: format '%S' expects argument of type 'wchar_t*', but argument 4 has type 'const char*' [-Wformat=]
wled00/../usermods/seven_segment_display_reloaded/usermod_seven_segment_reloaded.h:470:74: warning: format '%S' expects argument of type 'wchar_t*', but argument 4 has type 'const char*' [-Wformat=]
       sprintf_P(subBuffer, PSTR("%s/%S/+/set"), mqttGroupTopic, _str_name);
                                                                          ^
wled00/../usermods/seven_segment_display_reloaded/usermod_seven_segment_reloaded.h:470:74: warning: format '%S' expects argument of type 'wchar_t*', but argument 4 has type 'const char*' [-Wformat=]
*** [.pio\build\esp32dev\src\usermods_list.cpp.o] Error 1```

Can someone do a PR to add this or help point me in the right direction of what else I need to do to get this to compile? Thanks!

al3x1337, 2022-03-28:
Just discovered the 13.1 branch that as the updated usermods
https://github.com/atuline/WLED/tree/dev-v0.13.1-merge

al3x1337, 2022-03-28:
There's a bug within that branch:
[image]
The color wheel is missing and I can only use the CCT slider with RGB leds

closed 2022-03-28 by al3x1337

atuline, 2022-03-28:
Yep. Merging with AC's upstream master can often be challenging. We're looking into seeing what's wrong there.