atuline/WLED #190: 2D Geq Effect Varian

pnichols77, 2022-04-02:
**Describe the solution you'd like**
I  like the 2D GEQ graphic equalizer effect, but what i'd really like is a mirrored version - so the lowest frequency in the middle, with higher frequencies to each side, duplicated and mirrored. This would be like how old school graphic equalizers looked on stereos. Maybe they still do? I suspect it would be a fairly simple thing to do for someone who knows what they're doing (which unfortunately, when it comes to this, is not me).

ewoudwijma, 2022-04-02:
Good idea. I worked in it before and am happy to implement it. Will do next week!

pnichols77, 2022-04-03:
> Good idea. I worked in it before and am happy to implement it. Will do next week!

Another idea I had - for both the original and this possible variant - is that the colour schemes take effect horizontall rather than vertically. So, rather than each frequency being represented by differenrt colours in the scheme, instead it would go through the colour scheme as it increases in height. So, for instance, you could have green, amber and red, with the 'highs' of each frequency starting green, then amber in the middle, then red if they're peaking. 

Don't want to be making work for you just think it would be a cool way of introducing old school equalizer effects from 80s stereos onto something thats much more modern.

ewoudwijma, 2022-04-03:
Good idea as well, I will see what I can do, are you on WLED sr discord, so I can ping you if I have a question ?

pnichols77, 2022-04-03:
> Good idea as well, I will see what I can do, are you on WLED sr discord, so I can ping you if I have a question ?

Not really much of a discord user tbh but i can be quick to reply here - though ive just given you the bare bones of a couple of ideas im not sure how much help i could be!

ewoudwijma, 2022-04-07:
Hi there, I added your idea in a new version. See details here: https://discord.com/channels/700041398778331156/700701772838207640/961614510806933524

Currently you need to compile the dev-v0.13.1-merge branche to see this working. (This branch will be moved to the dev branch pretty soon)

pnichols77, 2022-04-15:
> Hi there, I added your idea in a new version. See details here: https://discord.com/channels/700041398778331156/700701772838207640/961614510806933524
> 
> Can you see this?
> 
> You need to compile the dev branch to see this working.
> 
> It looks pretty cool! Hope this is what you had in mind. If not let me know

Wow sorry totally forgot to check back in on this - really bad considering how quickly you worked on this. My weekend project is to work out how to get the built so I can try it!

ewoudwijma, 2022-04-15:
No problem! You can get builds here: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/Sound_reactive
Choose 0.13.1

pnichols77, 2022-04-15:
> No problem! You can get builds here: https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/Sound_reactive Choose 0.13.1

Sorry again for leaning on you but ive gone here https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/Sound_reactive/v013.1, downloaded and installed SR_esp32dev.bin

But I cant see any new effects. The build in hithub says its 2204120, but my wled is reporting build 2203150. Tried updating using the internal update method in WLED twice with no errors. Am I missing something?

ewoudwijma, 2022-04-15:
I see also 2203150 in my latest version. Where do you see the 2204120 in GitHub?

If you choose the 2D Centerbars effect and you see the following, then you have the right version:

[image]

If you play with the lowest 3 sliders (which act as toggles) you can see different modes of the graphic equalizer

pnichols77, 2022-04-15:
> I see also 2203150 in my latest version. Where do you see the 2204120 in GitHub?
> 
> If you choose the 2D Centerbars effect and you see the following, then you have the right version:
> 
> [image]
> 
> If you play with the lowest 3 sliders (which act as toggles) you can see different modes of the graphic equalizer (center vertical and color vertical are your ideas)

I was looking in the wrong place for both forget about it. That effect is amazing thank you - works great once i've tweaked the various sliders and just what I was imagining. 

Just one thing - the decay uses the colour of the 'bottom' of the display, rather than the top. So a bar that goes 'into the red' decays as a green bar. Would it make more sense for it to decay with the colour it reached?

ewoudwijma, 2022-04-28:
Submitted a new commit to dev where decay/peak colors are adjusted, looks more consistent now. Please take a look and tell me if this is what you had in mind

pnichols77, 2022-04-28:
> Submitted a new commit to dev where decay/peak colors are adjusted, looks more consistent now. Please take a look and tell me if this is what you had in mind

Ive just installed the latest dev build and it looks the same as before - do i need to wait a bit for it to include your changes? Sorry Im just not sure of the process/timescales.

ewoudwijma, 2022-05-10:
@pnichols77 , already checked latest version?

pnichols77, 2022-05-10:
> @pnichols77 , already checked latest version?

Hi sorry forgot to reply again! Yeah ive got it installed and it looks great. Thank you.

ewoudwijma, 2022-05-10:
Great! Then I close the issue

closed 2022-05-10 by ewoudwijma