atuline/WLED #194: SR Wled homebridge

lukaszwitkowski, 2022-04-08:
hello I add normal weed to my home bridge - apple HomeKit app and everything works great, now I want to add SR Wled to and is ok to change color or dimmer but I have problem with sound effects.
I wrote effect like that:
     "effects": [
                        "Rainbow Runner",
                        "Circus",
                        "Pacifica",
                        "Fireworks",
                        "Fire 2012",
                        "Rain",
                        "Police",
                        "Gravcenter"
                    ]
normal effect turn on and off but SR Wled don't work with command Gravcenter or else effects. What is name of sound effects
do you thing I need to add "music /sound icon" can you copy that for me here?

atuline, 2022-04-09:
You might want to join the Discord. I don't think any of us have played with anything bud the  web interface, but there are folks on Discord who have.

lukaszwitkowski, 2022-05-11:
Can you send me link? To group?

atuline, 2022-05-11:
That's shown on the home page of this repository at https://github.com/atuline/WLED

softhack007, 2022-07-08:
Hi @lukaszwitkowski, 

You can find the names of all sound reactive effects in this source file, starting at line 1403

https://github.com/atuline/WLED/blob/191d742036e05e0c7ea77f03d0d3ef892ddff0c6/wled00/FX.h#L1403

For example, the internal name of Gravimeter is " ♪ Gravimeter"  - including blanks and musical notes. Other examples are " ♫ Waterfall" or " ♫ 2D Akemi". Not sure if that's what homekit wants. But it's worth a try.

closed 2022-07-21 by ewoudwijma