atuline/WLED #198: soundReactive_WLED_0.13.0 for esp8266

Abedi98, 2022-04-13:
hi

When will soundReactive_WLED_0.13.0 be released for esp8266 (wemos d1 mini)?

Abedi98, 2022-04-13:
> hi
> 
> When will soundReactive_WLED_0.13.0 be released for esp8266 (wemos d1 mini)?

Version 0.12.1 does not have a playlist

tihoangyeudau, 2022-04-13:
@THATDONFC He will help you solved this with 0.13.1 version

dosipod, 2022-04-16:
> > hi
> > When will soundReactive_WLED_0.13.0 be released for esp8266 (wemos d1 mini)?
> 
> Version 0.12.0 does not have a playlist
Esp8266 13.0.b3 is here 
https://wled-install.github.io/

softhack007, 2022-06-18:
See issue #159.

README states that the ESP8266 branch is no longer receiving regular updates. Reasons explained in https://github.com/atuline/WLED/issues/159#issuecomment-1006640255 

Currently you need an ESP32 to run WLED-SR. Technicially the 8266 is not fast enough for our audio processing code, so it's not so easy to 'release th latest dev for esp8266'.

closed 2022-06-18 by softhack007

poelzi, 2022-08-16:
I want to use the reactive esp2866 as slave only, the master is esp32 with microphone. There is a sync option in the settings, do i understand those options wrong ?
Because as a slave, it should not matter if it is fast enough when the data is sent by the master ?

softhack007, 2022-08-16:
You might try that, but we cannot promise it will work when the "slave" is running an old version of SR WLED. The data formats have evolved and they might not be compatible with what older software versions are expecting to receive.

Even when "master" and "slave" are running the latest Version of SR WLED, sound sync is not very reliable and will produce frequent "stuttering" of your effects. Sound Sync might be OK with some effects like "freqwave" or "plasmoid". However the results are not the same as when the sound input (like I2S digital microphone) is directly connected.

See also 
* https://github.com/atuline/WLED/wiki/UDP-Sound-Sync#what-else-
* https://github.com/blazoncek/WLED/issues/37