atuline/WLED #199: WS2815 flickering at version 0.13.0-bx

razzz7, 2022-04-13:
### What happened?

Since updating on version 0.13.x my WS2815 LEDs are flickering.
Downgrading back to version 0.12.0 and everything works fine.

I am using a quinled dig-quad / ESP32 

### To Reproduce Bug

update on version 0.13.0-b3 and higher

### Expected Behavior

No flickering

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.0-bx

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

atuline, 2022-04-14:
Have you compared that to the exact same version of the original WLED?

razzz7, 2022-04-14:
Just installed Aircoookie version 0.13.1 and have no issues.
Back to sound reactive 0.13.b leds flickering.

atuline, 2022-04-14:
OTA updates maybe problematic because we've changed the file system. I'd do a fresh install:

https://github.com/atuline/WLED/wiki/Installing-and-Compiling#flashing-esp32-binaries

razzz7, 2022-04-15:
Thx for the hint. I did a fresh install according to this guide, but issue still appeared. Turns out I had to reduce the automatic brightness limiter a bit.

atuline, 2022-04-15:
If it's not one damn thing it's another. I've written pages of several things that can go wrong and that's just for FastLED animations. WLED takes it to a whole new stratosphere.

closed 2022-04-15 by atuline

razzz7, 2022-04-16:
Sry that I have to bring this issue up again.
I could narrow down the flickering.
After boot up/restart the leds are flickering. When I go to the led settings page and just hit save, without any changes, the flickering disappear and effects are normal.