atuline/WLED #201: 13.1 DEV corrupted FS

dosipod, 2022-04-16:
### What happened?

Using https://wled-install.github.io/ to install [SR13.1 DEV] results in corrupted FS . Using the same installer with normal wled is working fine . The same was tested on multiple esp32 versions .When saving presets serial monitor shows errors with LittleFS
[image]
 

### To Reproduce Bug

Use https://wled-install.github.io/ to install 13.1 DEV . Then create presets and reboot the esp, look at the serial monitor in ardunio IDE , you should see an error related to LittleFS

### Expected Behavior

As we are using the same installation method we expect srwled  fs to be fixed 

### Install Method

Binary from WLED.me

### What version of WLED?

13.1 DEV

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

atuline, 2022-04-16:
https://github.com/atuline/WLED/wiki/Installing-and-Compiling#flashing-esp32-binaries

dosipod, 2022-04-16:
SR13.1 DEV binary is only here https://wled-install.github.io/ , is this web installer  not supported ?
I am only testing multiple  units and if you can install SR13.1 DEV  from  https://wled-install.github.io/  without any issue then we can close this .

btittelbach, 2022-04-22:
not sure if this is related, but I observed that after upgrading to 13.1 my presets stopped working.
- presets where still there
- selecting a preset resulted in a red error message in the web-interface along the lines "connection to wled lost" each time.
- everything else (except analog microphone #162 ) worked fine

downgrading to 12.0 made presets work again.

TobiasJu, 2022-04-25:
@btittelbach just compiled 13.1 from the Master branch myself and yes presets stopped working. Most of the time selecting a preset does not change the animation and sometimes the esp32 resets itself.

softhack007, 2022-04-27:
What kind of micophone are you using? Analogue or digital? Which Pins?

Because there could be a relation to how the I2S driver is collecting samples, like watchdog reset or "WiFi dies" effects...

dosipod, 2022-04-28:
I am not sure if this is clear but we are testing multiple units and none of them are having any attached leds or microphone . 
Compiling from source and flashing with esptool as per the wiki also resulted in the same LittleFS behavior . Is it possible that this is related to my setup and not a bug ? 
It is still the case there is no issue with compiling or installing normal wled  from web installers on the same PC

softhack007, 2022-04-28:
Did you check serial output? In case you see some "guru meditation" or unexpected reset, that would be clearly a bug.

dosipod, 2022-04-29:
I am not really sure if the first picture I added is what you meant by check serial output because that is the  serial output errors but i am not sure if that is generic for all filesystem issues or it is pointing to something in the  code in lfs.c .

It would be very simple to close this if someone  could mention that  installing SR13.1 DEV from https://wled-install.github.io/ worked with no filesystem errors and give few details on their environment if esptool installed version even matters in such case  .

softhack007, 2022-05-26:
It seems that LittleFS has a known problem when used from multiple threads (missing semaphore). As consequence, you see bad blocks and temporary missing files. Maybe related ?

issue https://github.com/lorol/LITTLEFS/issues/44
PR https://github.com/lorol/LITTLEFS/pull/45

softhack007, 2022-05-26:
... I have created a fork of lorol LittleFS, and merged the pending PR that should improve thread-safety.
Please give it a try - you need to re-compile with the following change in `platformio.ini`:

find this in [esp32] section:
> `https://github.com/lorol/LITTLEFS.git`

and replace with
> `https://github.com/softhack007/LITTLEFS-threadsafe.git#master`

Please tell us if this helps.

dosipod, 2022-05-26:
Thank you very much , Would it be possible to replace the bad 13,1 Dev bin at https://wled-install.github.io/

softhack007, 2022-05-26:
I've just committed the necessary changes for platformio.ini into `dev` - see https://github.com/atuline/WLED/commit/212e2d3b8c2614baa02afbe16a57010f71f045d9

Now we need someone who knows how to upload binaries to wled-install....

dosipod, 2022-05-29:
we just tested https://github.com/wled-install/wled-install.github.io/blob/main/fw/SR_ESP32_4MB_13.1_DEV.bin [07d7dec]
on 3 different MCUs ( All of which worked fine with normal wled and SR13.0-B6 ) .

All our tests indicated the presets issues are  not resolved . Flashing the bin per the wiki  using esptool ,OTA or from web installer  resulted in the same issues with presets  .

[image]
[image]
[image]
[image]

jonasschwenk, 2022-05-30:
Same here. With SR 13 b6 everything works fine. After Upgrade to 13.1 presets are broken. No matter if installed ota or flash via esptool( inc erase ).

softhack007, 2022-05-30:
Some known problems in WLED that could be related to the presets problem:

- https://github.com/Aircoookie/WLED/issues/1981
- https://github.com/Aircoookie/WLED/issues/2681

The first problem affects the B6 prerelease, and could lead to corrupted json files that cannot be loaded by a newer version of WLED. The second one affects WLED 13.1 only, and could lead to presets not being loaded/saved.

Could you please post your your [WLED-IP]/presets.json files, before and after updating WLED? It might help us poor develops to better understand what happens on the devices.

dosipod, 2022-05-30:
Good News,  Members of the SR dev team are able to  reproduce the presets issues with 13.1 Dev.

ewoudwijma, 2022-06-27:
Softhack solved the preset issue. Can you confirm it is working for you?
I don't know status of corrupted FS, do we need to keep this issue open?

jonasschwenk, 2022-06-27:
I can also confirm its working for my setup. All 8 controller are now working with 13.1.  All proplems with the presets are gone. Excellent work and a big thank you.

closed 2022-06-27 by dosipod