atuline/WLED #203: battery monitoring/deep sleep when LEDs are off for Adafruit ESP32 Feather v2

pandaboy6621, 2022-04-23:
**Is your feature request related to a problem? Please describe.**
I am running a mobile instance of WLED on a hat and wanted to be able to monitor the battery percentage from the onboard lipo charger. I would also like my ESP32 device to enter deep sleep when the push button toggles the lights off to preserve battery power.

**Describe the solution you'd like**
I want a menu to be able to select for a battery monitor, according to https://learn.adafruit.com/adafruit-esp32-feather-v2/overview you should be able to use this data from "pin (GPI35) - This ADC pin can be used to read the data from the battery monitor. Basically perform an analog read and multiply by two, to get a rough sense of the lipoly battery voltage." I would also like a checkbox in the UI to enable/disable deep sleep when Leds are powered off. I am currently usign an Adafruit ESP32 feather v2 and a 3.7v 30,000mah lipo jst battery.

**Additional context**
I am currently usign an Adafruit ESP32 feather v2 and a 3.7v 30,000mah lipo jst battery.

ewoudwijma, 2022-04-24:
You want a lot 😉

atuline, 2022-04-24:
That actually sounds more like a usermod that could be used by regular WLED as opposed to SR WLED specific functionality. I recommend joining the regular WLED Discord and interacting with the community in order to get traction on this.

closed 2022-04-24 by atuline