atuline/WLED #204: Segments are gone after a reboot

XerXes777, 2022-04-29:
### What happened?

Segments are gone after a reboot

### To Reproduce Bug

Create Segments again
and after a reboot they are gone too

### Expected Behavior

Segments are gone after a reboot

### Install Method

Binary from WLED.me

### What version of WLED?

SR-WLED beta version 0.13.0-b6

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

atuline, 2022-04-29:
Did you read the Wiki page at:

https://github.com/atuline/WLED/wiki/Installing-and-Compiling

XerXes777, 2022-04-29:
I did these steps under the esp32 topic.

softhack007, 2022-05-11:
Did you create a preset that contains your segment settings, and did you configure WLED to load this preset at startup?
If not, maybe what you describe could be "expected behaviour"?

https://github.com/Aircoookie/WLED/wiki/Segments

>When you create segments be sure to save them as [Presets](https://github.com/Aircoookie/WLED/wiki/Presets) (along with any effect and palette/color selection) **otherwise they are lost if ESP reboots or looses power.**

softhack007, 2022-06-18:
Seems like this issue is actually expected behaviour. And it's not WLED-_SR_ specific.

closed 2022-06-18 by softhack007