atuline/WLED #205: Matrix modes seems to have a strange map, when using a stream to send something (ex. xLights)

lost-hope, 2022-04-30:
I have a Matrix with WLed SR on it and wanted to try out xLights, to display text. The shown picture looked really strange. When switching the WLed back to Strip mode, it works as intended. 

I would like to know or have a clear documentation about how the mapping of the streamed data should look like. 

I already asked about this on Discord, but wanted it to add here, so i can be tracked easier. 
Answer of ewowi:

> I worked last year on the 2D effects, but only on build-in effects to 2D matrix (function xy and logical2physical), not on e1.31, so might well be this is an untouched area ready for improvement. embedded-creations , any idea if there is some system or regularity currently or it is just an open area to be improved?

dosipod, 2022-05-04:
We did a lot of testing before with xlights and srwled and there is a lot combinations settings  in xlights and wled that could impact  how the effects are  displayed .

It might help if you provide more info on your setup and if you want us to  replicate your issue then please provide more  details on how  did  you setup e1.31 and universes  in xlights and your matrix info as  I  am not sure  of the  size but we have multiple 16x16 and 8x32 matrices  so hopefully we could match your setup

lost-hope, 2022-05-04:
So this is my physical led Layout. 
[image]

In WLED my Matrix settings are:
> **2D Matrix**
> Width: 32
> Height: 32
> **2D Panel**
> Horizontal Panels: 1
> Vertical Panels: 4
>**2D Panel Layout**
> Firstled Position: Bottom Right
> Orientation: Vertical
> Serpentine: true
> Transpose: false

The 2D Panel Layout looks not correct when comparing the picture above and the position of the first Led, but this is the only way i get a picture with it. 

My DMX Settings: 
[image]

In xLights i have just set up a controller and tried different models.

- a single line 1024Leds long and using start channel 1
- then a 32*32 Matrix with different starting Led locations and layouts
- an setup to mimic the real Layout. 

When using the single line you can really see the unusual Layout (also in the Peek on the WLED web gui). 
I can try to make a video of it later.

netmindz, 2022-05-23:
Possibly related to https://github.com/atuline/WLED/issues/175