atuline/WLED #209: 1D effects for music spotlights

roboraptor, 2022-06-08:
Would it be possible to add some effects for single diode spotliights and such? 
Effects like some on dmx lights auto or sound modes
- strobe on beat
- change color on beat
Thank yooou

softhack007, 2022-06-09:
Hi, currently we do not have specific effects for single (or very few) LEDs.
Hoewever most WLED effects also work on single LEDs, see screenshots below for some examples that might also work for you. They are for my "high power LEDs" segment which has only 3 lights.

I can imagine that some of your ideas could be implemented in SR WLED later, like
* use a trigger event to "shift" the color map,  or pick a new color map at random
* a possible trigger could be "change on beat", however our current beat detection would need some improvements - currently it's not good enough reliably detect "beats".

Hope this helps,
 Frank

[image]

[image]

roboraptor, 2022-06-09:
Hi, thank you for your answers.

For beat detection im currently waiting for I2C microphones to arrive because using ordinary mic with adc on pin36 wasnt really relayable??? i can probably rewire it to "line input" and send only bass from monitor output or rewire it more to implement bass crossover? so that could be better? 

The effect are really simple yeah. It is just a LED spotlight using RGB PWM output.
if you can add at least the "change color on beat" it would be awesome.

Also got me thinking if there could be any workaround like save the colors in presets and create playlist and switch in playlist on the beat ? 

Thank you very much
Robo

atuline, 2022-06-09:
Damian peckett had a nice beat detection article on his web site a while back, but unfortunately, that site is no longer available. Beat detection, however, is not an easy thing to implement, as is the automatic gain control.

I shlurped most of the information from that article, including the code if @softhack007 is interested.

roboraptor, 2022-06-09:
About the detection it is cool but I am more interested in the effect that I could use 😅
You know even the basic board in the fixture which detected beats was not perfect.

Yeah it is I2S INMP441

softhack007, 2022-06-09:
Btw, I think it would be a very simple exercise to create a "mood lighting" effect. 
Basically you need a prepared palette with the right "moody colors". A simple effect could cycle through the palette, and fill the complete strip with one of the colors. On next iteration, pick the next color from palette and repeat. 
This will work in "zero D" or with any longer strip, and even PWM LEDs.

roboraptor, 2022-06-21:
I tried the pallete.... After tedious hours of fighting with platformIO....
`{ 0, 255,  0,  0,
   36, 255,  0,  0,
   37,   0,255,  0,
   72,   0,255,  0,
   73,   0,  0,255,
  108,   0,  0,255,
  109, 255,255,  0,
  144, 255,255,  0,
  145,   0,255,255,
  180,   0,255,255,
  181, 255,  0,255,
  216, 255,  0,255,
  217, 255,255,255,
  255, 255,255,255};`
Just a basic one... But I am not sure about the effects...

softhack007, 2022-06-22:
Hey, nice to see that you have taken on the 'challenge' to create a new effect 😃

There are a few wiki pages that could help you get started:
- https://github.com/Aircoookie/WLED/wiki/Add-own-functionality#create-custom-effects
- https://github.com/atuline/WLED/wiki/WLED-Programming-Notes#updating-fxcpp

good luck, and happy hacking 👍

softhack007, 2022-06-22:
WLED-SR currently does not have a good 'beat detection' algo. Improvements are on our priority list, however this will happen after the next release.

We already have a variable `samplePeak` that has the  value '1' when the volume in one FFT bin (`binNum`) is above a predefined threshold (`maxVol`). You can look at the Waterfall effect (line 6496) for an example how to use it.

Defaults are set here. They can be changed by the effect for example using UI sliders (see below).
https://github.com/atuline/WLED/blob/95abf86a00ffe35d1cfb4a09a57f8483f3c4db56/wled00/audio_reactive.h#L57-L58

https://github.com/atuline/WLED/blob/95abf86a00ffe35d1cfb4a09a57f8483f3c4db56/wled00/FX.cpp#L6478-L6506

https://github.com/atuline/WLED/blob/95abf86a00ffe35d1cfb4a09a57f8483f3c4db56/wled00/FX.h#L1415

softhack007, 2022-06-22:
Just took at closer look at the "Palette" effect.
Now hat you mastered platformIO, you could try your first 'dirty hack' and modify the source code of `uint16_t WS2812FX::mode_palette()` in FC.cpp.

Try to replace the first few lines of the function - everything before `bool noWrap = ...` - with the code sniplet below. Then recompile & upload to your device.

Result: If you select the "Palette" effect and move the "Speed" slider to zero (or any low value < 32 ), then you can see how the colors only changes when a "beat" is ongoing. Speed > 31 will give you the normal, non-SR behaviour.

Cheers and happy hacking,
 Frank.

```` C++
uint16_t WS2812FX::mode_palette()
{
  uint16_t counter = SEGENV.aux0;
  if (SEGMENT.speed > 32)
  {
    counter = (now * ((SEGMENT.speed >> 3) +1)) & 0xFFFF;
    counter = counter >> 8;
  } 
  else {
    if ((samplePeak==1) && (sampleAgc > 1)) {
      counter += SEGMENT.speed +1;
      if (counter > 255) counter -= 255;
    }
  }
  SEGENV.aux0 = counter;

  // rest of the function remains unchanged ....
  bool noWrap = (paletteBlend == 2 || (paletteBlend == 0 && SEGMENT.speed == 0));
 ````

Of course that's just a quick try, but it might already be what you are looking for.
You can learn more about "persistent variables" here: https://github.com/atuline/WLED/wiki/Persistent-Variables-and-WLED#enter-wled---

softhack007, 2022-06-22:
>  Without me needing the proper setup and FFT? That would be awesome.

Actually it's just plain WLED - not LEDFx app for FFT.
So once you have changed the code, recompiled & uploaded, you can try it out.

Only setup needed is the basic sound settings in WLED. 
Squelch should be high enough so effects can detect silence (in my home, I use Squelch=8). 
Then set AGC to "Normal". That's all.

Test trigger: in a silent room, you can drop a few pencils onto the desk near the microphone, that should count as a "beat".

roboraptor, 2022-06-23:
OMG OMG OMG ❤️ 
I had to move the whole palette thingy under your sampleAgc definition but it looks like it works! 🎉 
Still need to wait for I2S microphones 🎤 
(Because the thing I have connected to pin 34 is only working when I blow on it or smack it with something...)

Also, you can set AGC more than just on/off ? I have only checkbox there.

roboraptor, 2022-06-25:
How do I know that I2S microphone is detected?
I cant make it wok...

softhack007, 2022-06-25:
We have a guide in our wiki, maybe this helps:

https://github.com/atuline/WLED/wiki/Digital-Microphone-Hookup

Actually you can now use the web UI for changing I2S pins, so the part that suggest to change `#define`s in source code is a bit outdated.

The INMP441 is very sensitive; I would start with gain=60 and squelch=8. If you have a LED strip (size don't matter), the 'gravimeter' effect should react once the mic is connected properly.

softhack007, 2022-06-25:
> Also, you can set AGC more than just on/off ? I have only checkbox there.

Please use source code from the 'dev' branch, don't  use 'master'. We have made many improvements for AGC in `dev` .

https://github.com/atuline/WLED/tree/dev

roboraptor, 2022-06-25:
> Please use source code from the 'dev' branch, don't use 'master'. We have made many improvements for AGC in `dev` .

Im so sorry... never used github much... Will try tonight ❤️ Thank you

roboraptor, 2022-06-25:
Im sorry but still not working... i bought 5 of them and none is doing anything... 
I even tried to remap pins to something different sck26 ws25 sd33 but still no avail...

softhack007, 2022-06-26:
Did you connect the "L/R" pin to GND, as described in our wiki?
If yes, then maybe you can try to put the pin to 3.3v, just to see if left/right are swapped.

Also it's worth to try and use the pins suggested in the wiki page, if these are available on your ESP32: sd=32 ws=15 sck=14

if still not working, you could join our Discord and ask in the "support" channel. https://discord.gg/4CQRmfR

If asking on discord, you may want to add a screenshot of your sound settings and a picture that shows how you wired the mic to your ESP32.

softhack007, 2022-06-26:
You can also compare your setup to this: https://github.com/atuline/WLED/issues/184#issuecomment-1073976540

softhack007, 2022-07-01:
Does it work now?

softhack007, 2022-08-12:
Closing, as no activity for some months now.

"Zero D" effects are nothing special - any effect that works with "PWM LEDs" is actually "0D" as the same color will be shown on each LEDs connected to PWM pins.

closed 2022-08-12 by softhack007

netmindz, 2022-11-30:
@roboraptor you might also want to check out the audio responsive color palettes I've added to the MM fork