atuline/WLED #210: Suspend intensive calculations when LEDs are off

B0rax, 2022-06-17:
I have WLED SR lamp as a desk lamp which is connected to power 24/7.

I couldn't help but notice that the ESP32 gets quite hot even when the LEDs are off. I also measured the power draw because of this.
The ESP32 seems to be using 200mA while displaying nothing. This is 1W standby power which seems quite high to me. This is almost 9 kWh per year.

Maybe it would be possible to drastically reduce this power consumption. I think this would be a great enhancement.

softhack007, 2022-06-17:
Good observation. It is true that WLED-SR currently is never idle, even when LEDs are inactive.

However the device is actually never in stand-by:

1) For the SR part, the microphone is sampled continuously, and sound analysis (filtering and FFT) also run all the time. We want to react on sound, so we have to keep checking for sound. FFT currently occupies one of the two cores completely, at 240Mhz...

2) All other possible inputs (buttons, USB, network, etc etc) are also being checked regularly.

3) Another well-known "power eater" is the WiFi connection. Wifi is actually responsible for a large part of ESP32 power consumption, it can go up to 500mA.

Currently there are not plans to implement a "low power" mode in WLED-SR. The main reason is that the typical "300LEDs strip" consumes a lot more power than the ESP32, even when all LEDs are off.

My suggestion for your temperature problem would be to start looking for cooling pads that can be put directly onto the chip; many are intended for "Raspberry Pie" controllers, however they might also fit onto the ESP32. If you still have too much heat, try cooling the board with a small fan. 

We might look into the "low power" mode for a future version, however honestly speaking this is not on out priority list.

cheers,
 Frank.

softhack007, 2022-06-17:
Like these ones (amazon Germany):
* small aluminium heat sink set: https://www.amazon.de/CTRICALVER-Aluminium-Heatsink-Cooling-Radiator/dp/B08YN4VGXN
* small USB fan, 4cm: https://www.amazon.de/dp/B07F1NV9T7

B0rax, 2022-06-18:
Thanks for your reply, I understand that power consumption while displaying anything is not a concern.

As far as I understood it, the most intense calculations are the FFT analysis. Is it possible to pause these calculations when the LEDs are switched off or (even better) when no sound reactive animations are displayed?

If I read the code correctly there is already a place in audio_reactive.h where the FFT calculations are paused when sync is enabled:
`    // Only run the FFT computing code if we're not in Receive mode
    if (audioSyncEnabled & (1 << 1))`

maybe a additional condition could be placed here?

Again, I am not talking about reducing power consumption or heat while it is running, I am only concerned with the power draw while nothing is done in a 24/7 application.

softhack007, 2022-06-18:
Hi 
The code line that you point to might be a start, however it would also disable sound sampling because the FFT thread first pull new samples from I2S, then runs the FFT analysis. 

As experiment to understand how much power can be saved, you could 
* comment out the `if(...)` condition, so that `continue;` is always executed. 
* Re-compile & upload to your device.
* Then measure the power consumption again. I would be interested to hear if you really see a reduction in power consumption.
* After measuring, please un-do the change, as it makes WLED-SR ignore all sound inputs. 
* Re-compile & upload again so you have a working WLED-SR.

For a clean solution, can you specify what "nothing is done in a 24/7 application" means for you? 
Is it about running a sound-reactive effect just with brightness set to zero? Or a non-SR effect that set a specific color? Turning the device off ?  Are you expecting WLED to still respond to switches, timers etc when "nothing is done"?

softhack007, 2022-06-19:
Just looked into the ESP32 datasheet from espressif (https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf). 

It seems that your measurement of 200mA is already a low value when Wifi is active. 

[image]

Generally speaking, there are ways to reduce power consuption:
* actively put ESP32 into sleep mode (many modes availeable)
* put wifi radio into light sleep
* reduce chip frequency (160Mhz or 80Mhz)

AFAIK, none of these is currently implemented in WLED. And for the typical use case, none of them would not make much sense.

There are many guides on ESP32 power saving on the internet, like this one https://www.mischianti.org/2021/03/06/esp32-practical-power-saving-manage-wifi-and-cpu-1/ . However none of them seems to suggest that keeping the chip idle (executing delay()) will help. Maybe embedded processsors are different from Desktop Computers  in this respect...

.

Maybe you can create a usermod that puts the chip into sleep, reduces wifi power and cpu frequency - based on your own criteria for "nothing is done". Not sure if there are WLEDusermod available already for power saving.

blazoncek, 2022-07-06:
@softhack007 perhaps look into how I added on/off button to AudioReactive usermod.

closed 2022-07-14 by softhack007