atuline/WLED #211: Intermittent Boot loop

jwardle, 2022-06-22:
### What happened?

Hi, 

My AzDelivery ESP32-WROOM-32D intermittently enters into a boot loop - sometimes it can be 20min after initial bootup, sometimes its 4 days after bootup. I am using the ESP32 with WLED-SR v0.13.1 currently, connected to a 2 metre string of 120 (total) WS2812 LEDs, also with a DollaTek INMP441 microphone for sound recognition. The lights and ESP32 are powered by a 5V 9A adapter - with the ESP32 powered off the 5V pin. I do have the WLED configured with a software limit of 4800ma currently.

The issue is, WLED mostly fine before the boot loop. The lights do reboot intermittently when  on a static off-white glow (without any external influence - not changing a preset, seen via the uptime in the info panel) however less concerned about this. If I turn the power off and on when the issue occurs, the bootloop continues, however If I turn off the LEDs at the power, and leave them for a while before turning the lights back on: they work as expected for a period until it next happens. 

I had thought this may be dodgy wiring, or overheating of the ESP32 (the assembled unit is housed in a plastic sealed box) however have checked the wiring, and run the device in the open - the same effect is seen. 

When the ESP32 reboots, the lights are on for ~1 sec and i've noticed that some of the LEDs at the end of the strip are sometimes a blue colour, whilst all others are the intended white colour. The length of the blue LEDs varies, but always starts at the end of the strip (opposite to end connected to gnd/data/5v).

I am unsure what the issue could be at this stage.

### To Reproduce Bug

Chip: ESP32-WROOM-32D
Mic: INMP441
LEDs: WS2812 60 LED/m x 2m

Power the LEDs and ESP32 via 5v adapter. Leave lights on a white glow, full brightness with a 4800ma limit. After a period the lights/ESP32 enter a bootloop. 

### Expected Behavior

Lights do not boot loop.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2022-06-28:
Yes the v0.13.1 version you installed is not working correctly , for the time being try to install SR13.0-b6 from web installer  . Or you  can test the latest v0.13.1 dev  from here https://wled-install.github.io/   ( There has been reports of e1.31 issue but no reboot or issues with presets ) .

jwardle, 2022-06-29:
Unfortunately I’ve tried SR13.0-b6 and experience the same issues. Will try latest dev version via the web installer.

dosipod, 2022-06-29:
I hear this name "AzDelivery " a lot . Did you by any chance try to install normal wled and it worked on this MCU ?
I have setups of around 10 all running stable SR13.0-b6 at the moment   installed from here https://install.wled.me/ 

I am doing only testing but if you are after a solution to an actual issue then try to visit the discord

softhack007, 2022-07-06:
> I hear this name "AzDelivery " a lot.
They are a Germany- based reseller of Chinese Arduino-related hardware, offering better support. They put their "AZDelivery" mark on the hardware, but usually inside you find a vanilla design like AiThinker, WEMOS etc. https://www.az-delivery.de/en

Very well-known here because their are selling on Amazon.

softhack007, 2022-07-14:
> Again, turning off for a period results in the device working normally for a period before boot looping.

Could you connect USB serial to a PC, and check for any messages on serial monitor (You could use one one from Arduino IDE)? Your "boot loop" sound a bit like under voltage, or maybe a restart due to "brownout detected".

dosipod, 2022-07-19:
Assuming the MCU is working correctly  ,please test the latest version 0.13.2-a0 (build 2207171)   https://wled-install.github.io/ this seems very stable in our tests

jwardle, 2022-08-12:
I've managed to finally get hold of UART USB adapter and connect it to the lights and capture the logs when the lights reboot. @softhack007 is correct that the brownout detector is being triggered (log below). 

I do have a 1000mf capacitor across the 5V in (to ESP32 and LEDs) and GND following advice from WLED discord users. Do need another/something else?

I do notice that the DC adapter is getting extremely hot, to the point it's too hot to hold. Is this something to be concerned about, or related?

`Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5360
entry 0x4008067c
AS: Generic I2S Microphone - left channel only.
Ada`

softhack007, 2022-08-13:
> I do notice that the DC adapter (the female head wires from ESP32/LEDs plug into, not the adapter brick itself) is getting extremely hot.

Oh-oh, yes this is not "normal". 

Maybe you are using some screw terminal adapter like this?
[image]
These parts are no good if you need more than 3 amps. You should try a different kind of connection. 

It might be better to strip the wires, terminate them with [a wire ferrule](https://blog.delcity.net/what-is-a-wire-ferrule/) of the appropriate wire gauge, and use screw terminals to make the connection.

You may want to join our discord
* SR (sound) WLED: 
[image]

* WLED generic: 
[image]

jwardle, 2022-08-16:
Thanks. I’ve replaced the power supply unit with a new one from a different vendor… the lights are now stable. Closing this issue. Not a bug, but faulty hardware it appears!

closed 2022-08-16 by jwardle