atuline/WLED #213: very slow LED response with LedFx (e1.31 protocol) for both 2D, 1D LED config, faster on WLEDSR v.0.12.0

christiansuryanto, 2022-06-25:
### What happened?

WLEDSR displayed embedded patterns smoothly.

Tried to perform patterns from LedFx, properly connected to WLEDSR via E1.31.
Checked Wifi, LED, properly set.
Slow display response happens on both 1D and 2D LED config.
Restarted the WLED, no fix.

WLEDSR 0.12.0 has faster response on displaying patterns from E1.31 (LedFx), for both 1D and 2D LED mode.

Link for videos :

- [WLEDSR_0.13_2D](https://youtube.com/shorts/QWJ7JUywQuA?feature=share)
- [WLEDSR_0.13_1D](https://youtube.com/shorts/dpLkA-EZ0HU?feature=share)
- [WLEDSR_0.12_1D2D](https://youtube.com/shorts/XF_9KEpWxJQ?feature=share)

### To Reproduce Bug

Setting up WLED as from erasing flash, LED, Sound, until connected to local network properly.
On PC, setup LedFx, runs normally, add WLEDSR (using autodetect mdns and manual IP, same results on slow leds pattern display).

Try to change LED config 1D/2D, restart WLED, same results.

### Expected Behavior

on WLEDSR 0.12.0, display patterns from E1.31 protocol displayed faster, both 1D or 2D LED config.
(I think fps more than 20).

### Install Method

Binary from WLED.me

### What version of WLED?

soundReactive_WLED_0.13.1_ESP32

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Manut38, 2022-06-26:
I have this exact same issue.
Together with the preset saving/loading issue version 0.13.1 is pretty much unusable to me for now.

ewoudwijma, 2022-06-27:
E1.31 is still an open issue. 
The preset saving/loading has been resolved in 0.31.1 so hopefully that is no longer an issue.

dosipod, 2022-06-27:
Yes presets saving/loading no longer crashes the MCU  after the update 3 days ago . The issue with e1.31 and udp has been reported by multiple guys , we have tested 6 units from our side and they all became very slow to  none responsive  when used with e1.31 from applications like ledFx or Jinx  .

netmindz, 2022-07-03:
Given the webui shows when you are receiving e1.31 data, I wonder if there are parts of the code that could be suppressed if you are in "dumb slave" mode? Early return for pattern rendering, audio processing etc?

ewoudwijma, 2022-07-05:
I am trying to reproduce the error on my M1 Mac but get errors installing LedFX. Any suggestions what to do? Is there an alternative which runs smoothly on Mac which produces e1.31 data?

softhack007, 2022-07-05:
Just a wild guess - could there be a conflict with DMX? 
DMX is (beasicially) another serial protocol, and upstream WLED has introduced a replacement because "ESPDMX did not work on ESP32". https://github.com/Aircoookie/WLED/pull/2652

softhack007, 2022-07-07:
I just tried a simple "gradient" effect with latest WELD-SR `dev` and 150 pixels strips. I see it running at 30-40FPS right now.
So maybe we need more information about your setup to help us reproduce the problem.

@christiansuryanto, @dosipod : could you upload a screenshot from your "Device Settings" and "Effect Control" page in LedFx?

christiansuryanto, 2022-07-08:
> I just tried a simple LedFx "gradient" effect with latest WELD-SR `dev` and 150 pixels strips. I see it running at 30-40FPS right now. So maybe we need more information about your setup to help us reproduce the problem.
> 
> @christiansuryanto, @dosipod : could you upload a screenshot from your "Device Settings" and "Effect Control" page in LedFx?

[image]
[image]
[image]

I attach 3 screenshot files :

- LedFx Device management main page
- LedFx Device management Detail page
-Ledfx effect control page, I choose gradient, and WLED SR displayed on low fps (gradient chages displayed not smoothh. Color gradation is good, but brightness changes is not varies smoothly).

I use this binary :
[SR_ESP32_4MB_13.1_DEV.bin](https://raw.githubusercontent.com/wled-install/wled-install.github.io/main/fw/SR_ESP32_4MB_13.1_DEV.bin)

blazoncek, 2022-07-08:
@christiansuryanto can you test experimental version as well? It is supposed to get merged upstream.
If you do not like pre-compiled binaries use my fork and checkout `2D-support`. Add `-D USERMOD_AUDIOREACTIVE`in your `build_flags` and use `https://github.com/blazoncek/arduinoFFT.git` in your `lib_deps`. This will add audio-reactive capabilities. There is an option to switch on and off audio in Info page so you can test with and without audio enabled.

christiansuryanto, 2022-07-08:
> @christiansuryanto can you test experimental version as well? It is supposed to get merged upstream. If you do not like pre-compiled binaries use my fork and checkout `2D-support`. Add `-D [...]

unfortunately I am still unable to acomplish compiling & bulding binary myself..
Im not using platformio yet..

softhack007, 2022-07-08:
🤔 hmm ... The only main difference (in LedFx) between my setup and your (slow) setup seems to be that I have connected my WLED device by local IP address, while you are using its name "wled-chris.local".

I can imagine that the slow-down is due to DNS name resolving. Could be that _each_ packet from PC to WLED goes through DNS, meaning that your PC asks the Wifi gateway "where is `wled-chris.local` ?" about 30 times per second. That might explain the slow-down.

<br/>

--> Just to exclude that the problem is DNS related, could you try what happens when you attach to WLED by IP address, instead of using the name? See my example below.

[image]

If your strip reacts faster when using a numerical IP address, we would have an idea where to look for further analysis. If no change, then we have at least excluded one possibility.

christiansuryanto, 2022-07-08:
> > @blazoncek unfortunately I am still unable to acomplish compiling & bulding binary myself.. Im not using platformio yet.. Is there any binary available for experimental version? Did you mean binary from Atuline/WLED with version of 0.13.0bx alpha ?
> 
> https://github.com/srg74/WLED-wemos-shield/tree/master/resources/experimental

@blazoncek I use this binary [[WLED_0.14.0-bl0_ESP32.bin](https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/WLED_0.14.0-bl0_ESP32.bin)](https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/WLED_0.14.0-bl0_ESP32.bin) from your link of experimental fork, and the result is WLED animation runs smoothly on data from LedFx (e1.31 protocol), both for using numerical IP (192.168.100.116) and mDNS ip (wled-chris.local) on LedFx Device Management, and both using 1D and 2D matrix configuration also resulting fast performance on displaying effects from LedFx.

I notice that the [WLED_0.14.0-bl0_ESP32.bin](https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/WLED_0.14.0-bl0_ESP32.bin) binary not implementing / activating Sound Reactive feature yet, is there any experimental binary that already implement SR?

blazoncek, 2022-07-08:
> I notice that the [WLED_0.14.0-bl0_ESP32.bin](https://github.com/srg74/WLED-wemos-shield/blob/master/resources/experimental/Firmware/WLED_0.14.0-bl0_ESP32.bin) binary not implementing / activating Sound Reactive feature yet, is there any experimental binary that already implement SR?

I am calling it audioreactive and it is implemented as a usermod. So go to Config / Usermods and configure it there, I do understand that @srg74 compiles with usermod enabled.
It should also show in Info dialog if and when it is running.

softhack007, 2022-07-10:
@christiansuryanto, @dosipod : It might be a problem due to high processor load.

To verify my idea, could you enable "UDP Sound Sync" in receiver mode on your WLED device? Then restart your device, and try again to use LedFx. Do you see higher FPS now? Afterwards please set "sound sync" back to disabled.

To enable sound receiver mode, go to "Config" -> "Sync Interfaces", scroll down, and change to:
[image]

Then save, and restart  your device. 
For this test you don't need a second SR WLED in "Transmit" mode.

Technically speaking, when SR WLED is is sound receiver mode, then all sound processing (including FFT) is suspended. 
If you get better FPS when in "sound receive mode", that means my idea will work and I can make a bugfix in SR WLED.

christiansuryanto, 2022-07-10:
I have tried the Receive mode, and it indeed improves the visual performance of WLED regarding this issue.

christiansuryanto, 2022-07-14:
@softhack007 Thank you, Frank. Already discussed in discord, tried the firmware ([firmware_soundReactive_esp32dev_dev2207110.zip](https://github.com/atuline/WLED/files/9084483/firmware_soundReactive_esp32dev_dev2207110.zip)), and it went very well solving this issue.

softhack007, 2022-07-14:
Thank you, too. 
Please simply hit "close" when it's OK for you now.