atuline/WLED #220: make custom effect scripts AGC aware

softhack007, 2022-07-26:
It seems that "custom effect" WLED scripts are not getting the AGC sample, so they can only see the non-AGC version.

https://github.com/atuline/WLED/blob/1eb97c36ba3f7b67343529812a3e512d5ca1adc2/wled00/src/dependencies/arti/arti_wled.h#L24-L25

https://github.com/atuline/WLED/blob/5a6b5c66aa73e480f545cba0d74023a6df098413/wled00/src/dependencies/arti/arti_wled.h#L368-L369

I think this can be changed easily to 
`return ( (soundAgc) ? sampleAgc : sampleAvg);`

with the benefit that existing scripts do not need any changes.

Optionally, we could make the "unfiltered sample" availeable to custom effects. The "right one" can be selected this way:
```C++
  int tmpSound = (soundAgc) ? rawSampleAgc : sample;
```

ewoudwijma, 2022-07-26:
Perfect fix. Propose to do the optional later, also because I don't know the difference between filtered and unfiltered 😜

softhack007, 2022-07-28:
Solved.

closed 2022-07-28 by softhack007