atuline/WLED #221: Potential Bug: disabling digital mic in settings NOT disabling the mic

cr4nkz, 2022-07-26:
### What happened?

Hello everyone, i think i've encountered a bug maybe: I have got a board with analog aux in (3.5mm jack) and also a ICS43434 Mic onboard. The mic works flawlessly, but it seems like i cant stop the mic from working by disabling it in settings. Also, it seems like, the analog input (GPIO36) will not be listened to, if the digital mic is enabled. Please note: I switched the WS Pin from GPIO15 to GPIO22, maybe thats the Problem?

### To Reproduce Bug

Try to use a ICS43434 digital mic on Pins: SCK=GPIO14, WS=GPIO22, SD=GPIO32. In my case, the mic always sends informations, cant be disabled.

### Expected Behavior

I thought disabling the digital mic will prefer the analog input (GPIO36) for controling the LEDs.

### Install Method

Binary from WLED.me

### What version of WLED?

0.12.0a

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-07-26:
Just a few inital thoughts, as I did not verify your report in detail yet.
> analog input (GPIO36) will not be listened to, if the digital mic is enabled.

This is expected behaviour. SR WLED will only use one sound input source at any time, i.e. the one that is selected in [sound settings](https://github.com/atuline/WLED/wiki/Sound-Settings). No need to set analog input to "-1" if you don't use it. Please make sure that you reboot your ESP32 (reset key) after changing sound input settings. 

> Version: 0.12.0a

This is very old software already. Could you try our [current `dev` version](https://github.com/atuline/WLED/tree/dev), to confirm that the problem still exists? In case you cannot build the software yourself in platformIO, there are firmware images availeable on several sites, see https://github.com/atuline/WLED/wiki/Installing-and-Compiling#unofficial-development-binaries---here-be-dragons

softhack007, 2022-07-26:
Just for my curiosity, 
what is your motivation for trying an analog MIC instead of the digital one? 
Usually digital MICs work _a_lot_ better that analog input.

cr4nkz, 2022-07-26:
> Just for my curiosity, what is your motivation for trying an analog MIC instead of the digital one? Usually digital MICs work _a_lot_ better that analog input.

Thanks for your quick answer. I already reboot (Power down) my Esp32 after pin change. I use both, analog and Digital. Digital Mic is ics43434 and the analog Option is Aux in with my Board. I thought after disabling the Digital mic, i can use the analog in as a Backup. I will try the dev build as soon as possible and come back :)

softhack007, 2022-07-26:
> I thought after disabling the Digital mic, i can use the analog in as a Backup.

Actually that feature never worked reliably, because we actually cannot "detect" if a digital MIC is present on I2S.
The current software uses whatever you configure in the settings. On USB Serial, you'll get a error message in case there is a problem with the GPIO you want to use.

cr4nkz, 2022-07-26:
> > I thought after disabling the Digital mic, i can use the analog in as a Backup.
> 
> Actually that feature never worked reliably, because we cannot really "detect" if a digital MIC is present on I2S - its an uni-directional bus so the only detection would be that samples != 0 arrive on the configured GPIO...
> 
> The current software uses whatever you configure in the settings. On USB Serial, you'll get a error message when there is a problem with the GPIO you want to use.

Thank you for the answer! So whats the way for me to go, if i want to use both, but not at the same time?

softhack007, 2022-07-26:
With the latest software, you can first configure analog mic input (using any gpio related to ADC1), reboot, and check that it works. Then xhange sound setting to use your ICS43434 digital mic, also any GPIO that is available. 
Settings for the analog mic will still be saved as part of your config. When you want to use analog, change the audio input in sound settings.  

Also make sure to configure squelch, then turn on AGC. Unfortunately "squelch" and "gain" are global and not per-device, so you have to re-adjust them whenever you change the sound input.

Not sure how that works in older versions, though.

cr4nkz, 2022-07-26:
Problem can be closed. Its fixed with recent build. But maybe consider to implement independent values for analog / digital sound inputs. Thank you very much!

softhack007, 2022-07-26:
You're welcome. Good to hear that latest build works for you 👍
Right, having "squelch" and "gain" per-device might be a good enhancement.

closed 2022-07-26 by softhack007