atuline/WLED #225: improve beat (onset) detection

softhack007, 2022-08-12:
As part of the discussion in other problem reports, users have asked for beat detection. This issue ticket is meant to collect availeable information that might be useful for a future implementation.

Actually I already have a working prototype for beat detection based on the "beatroot" algorithm. However there might be other ways to achieve the desired function.

@atuline wrote in https://github.com/atuline/WLED/issues/209
> Damian peckett had a nice beat detection article on his web site a while back, but unfortunately, that site is no longer available. Beat detection, however, is not an easy thing to implement, as is the automatic gain control.
> I shlurped most of the information from that article, including the code if @softhack007 is interested.

Please add any ideas, expectations or anything that might be related.

Drizzt321, 2022-12-07:
Hmm...I presume you're talking about the URL mentioned [here on Adafruit](https://blog.adafruit.com/2015/04/07/beat-detection-on-the-arduino/), which is http://dpeckett.com/beat-detection-on-the-arduino. Through some Wayback Machine sleuthing, I think https://web.archive.org/web/20181202114804/http://damian.pecke.tt/beat-detection-on-the-arduino might be the post you're looking for?

softhack007, 2022-12-07:
Hi, i was actually thinking about something FFT based. However your links are also interesting 👍Let's see what the result will be. 
I actually have a prototype already that is based on recommendations from the "beatroot" science papers. It is an approach without filtering - computing the "spectral flux" over time from FFT results, and then applying some heuristics to find major peaks in the flux.
Maybe your references will add something special to my final implementation, so really thanks sharing 😃

Drizzt321, 2022-12-07:
@softhack007 You're welcome! Did nothing but use archive.org to help find a copy of a post mentioned above which sounded like it'd help.