atuline/WLED #227: Everything syncing except white balance

conkydong, 2022-08-30:
### What happened?

Everything is syncing except the white balance. If I the lights are set to cold light and I change them to warm and vice versa only the light I'm controlling is changing.

### To Reproduce Bug

See problem.

### Expected Behavior

See problem.

### Install Method

Self-Compiled

### What version of WLED?

SR WLED 13.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

conkydong, 2022-09-01:
Help?

softhack007, 2022-09-02:
The sync feature of SR WLED is the same code as "upstream" WLED by Aircoookie.
Please open a new issue ticket in Aircoookie/WLED here: https://github.com/Aircoookie/WLED/issues.
If your problem gets solved there, we will get the fix automaticially, the next time we sync our codebase with "upstream".

There is some general information about the "sync" feature in WLED knowledge base:
https://kno.wled.ge/interfaces/udp-notifier/

closed 2022-09-02 by softhack007