atuline/WLED #231: Max 4466 Mic won't work in 13.2

ghpk, 2022-09-15:
### What happened?

I took a fresh ESP32 Board,
Flashed Wled suggested bootloader, then flashed the 13.2  but microphone wont respond.

I had another ESP32 and new  MAXX4466 Microphone,
build up the circuit again on another PCB and it wont react to any sound, WLED static effects would work just fine.

Sound Reactive wont work No matter what gain I select,
infact I've made this earlier too and quite comfortable with tinkering settings but it won't work for sound.

### To Reproduce Bug

Flash esp32 with currently available latest build and use MAXX4466 microphone it wont work, static effects would work fine though.

### Expected Behavior

Wled should react to music.

Same esp32, I flashed SR WLED 0.13.1 and it works fine.
WLED 0.13.1 (build 2203150)

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.2  (2207281)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

CnLm3d, 2022-09-15:
esp 32 devkit sahibiyim. max 9814 mikrafon ile ben de aynı sorunu yaşıyorum. 36 nolu pinden ölçüm yaptığımda 1,2V ile 1,4V arasında ölçümler alıyorum ancak şans eseri çalıştırsam bile açıp kapattığım anda ses efektleri çalışmıyor.

softhack007, 2022-09-17:
In general the MAX4466 does work, but it is very sensitive to voltage fluctuations so not exactly the best device to use.

Did you check that the GPIO is set correctly in sound setting?
https://github.com/atuline/WLED/wiki/Sound-Settings#analog-input-pins

Labersack99, 2022-09-18:
I have the same problem with 3 different Devices, all ESP32 with MAX9814 (GPIO36, Gain+VCC connected).
Tried with Gain, Squelch and AGC setting, nothing worked.
Flashed version 0.12.1-b1 and mic in all three devices is working again.

softhack007, 2022-09-18:
Please check if it works when using another GPIO for the microphone input, like GPIO32.

ghpk, 2022-09-18:
No, It doesn't considers any GPIO's
Tested on both 30 PIN and 38 PIN ESP 32 boards.

thanks,

softhack007, 2022-09-18:
I remember that some users found a "magic trick" that sometimes helps when analog mics refuse to work.
It involves installing SR13.2 via usb from here https://wled-install.github.io/ ( only usb from that site and no OTA or other method). @dosipod maybe you have some more ideas?

Labersack99, 2022-09-18:
I tried installing via the WEB-GUI  https://wled-install.github.io/ .
It's the same as installing via OTA, WLED-SR is working, but no SoundReactive effects.

softhack007, 2022-09-19:
hmm 🤔 there must be something special about your ESP32...Analog Mic definitely works for many people using 13.2
SR-WLED 13.0-b4 did _not_ use i2S drivers for analog inputs, but was based on analogRead().

Could it be that you have an ESP32 "revision 0"? 
Revision 0 has many known hardware-bugs, which could explain the problem you see.

You can easily check your HW revision with a simple Arduino sketch:
```C++
#include <Arduino.h>
#include <Esp.h>

void setup() {
  Serial.begin(115200); delay(2000); Serial.println();

  Serial.print("ESP chip model   : "); Serial.print(ESP.getChipModel());
  Serial.print(", Revision ");         Serial.println(ESP.getChipRevision());
  Serial.print("CPU Speed (Mhz)  : "); Serial.println(ESP.getCpuFreqMHz());

  Serial.print("FLASH size (kb)  : "); Serial.println(ESP.getFlashChipSize() / 1024);
  Serial.print("FLASH Speed (Mhz): "); Serial.print(ESP.getFlashChipSpeed() / 1000000);
  Serial.print(", Mode ");             Serial.println(ESP.getFlashChipMode());

  Serial.println(); Serial.flush();
}
void loop() {
    delay(2000);
}
```
To run this, you only need Arduino IDE and the "ESP32 Arduino" board package. Compile & upload, then open Serial Monitor and press "RST" on ESP32.  

On my ESP32, I get this output:
```txt
ESP chip model   : ESP32-D0WDQ6, Revision 3
CPU Speed (Mhz)  : 240
FLASH size (kb)  : 4096
FLASH Speed (Mhz): 80, Mode 2
```

Please post the output you get.

wgtp, 2022-09-19:
The buttons are disabled. I think this was suggested in https://github.com/atuline/WLED/issues/162
[image]

Here is my output. Looks to be revision 1.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13160
load:0x40080400,len:3036
entry 0x400805e4

ESP chip model   : ESP32-D0WDQ6, Revision 1
CPU Speed (Mhz)  : 240
FLASH size (kb)  : 4096
FLASH Speed (Mhz): 80, Mode 2

softhack007, 2022-09-20:
Chip info looks good. 

Sorry don't have additional ideas as to what else can cause your problem. I've cross-check related source code, and everything that's related to analog input looks  good.

There are some generic clues in our wki that helped others make theit analog microphones work better. Maybe you find something here: https://github.com/atuline/WLED/wiki/First-Time-Setup#noise-and-spikes

wgtp, 2022-09-20:
Thank you for looking into this problem. I have a spare esp32 that I might start fresh and see what happens. If that doesn’t work, I will just get a digital Mic.

nielsaust, 2022-10-06:
Hi there. Any updates on this? I am facing the similar problem. I'm using a quinled-dig-uno v3 with a max4466 mic. The leds respond in SR mode but doesn't seem to be related to the mic, because they also light up in complete silence. Unplugging the mic also still makes the light bounce, so it seems I cant fault the mic(?).

Any ideas?

lorenz24, 2022-10-14:
Same problem, but if I reset the module the micro is working. 
Only cold start causes the problem.

lokster, 2022-10-22:
I can confirm the same issue happening for me as well - cheap ESP-WROOM-32 (38 pin). None of the suitable analog inputs work.
0.12.0 works fine, 
v0.13.0-b4 works fine, 
v0.13.0-b6 and newer - do not

softhack007, 2022-10-22:
0.13.0-b6 is the last version that uses analogread() for taking mic samples. All never version are based on I2S-ADC.

Please check that you have not configured any "analog button" (potentiometer) in WLED, and you don't use any usermods that call analogread().

There is a known bug in espressif software when using I2S-ADC together with analogread().

lokster, 2022-10-22:
Does that mean that after version 0.13.0-b4 it is not possible to directly send analog signal to the esp32 (for example from Line Out) and must use additional hardware?

softhack007, 2022-10-22:
> Does that mean that after version 0.13.0-b4 it is not possible to directly send analog signal to the esp32

No, the situation is not *that* bad. Actually there are many users who are happy with SR 0.13.2 and analog input (mic or line-in) and they never experienced any problems. This is Including myself- I still have one ESP32 here with an analog microphone, and my setup works since month.

Then, there are cases like yours, where the same software does not work reliably with the same analog input devices. We never understood the difference, but it's probably something deep inside the hardware drivers that we cannot change.

The known bug from espressif is relevant when you have analog audio (mic or line-in) **and** another analog signal, for example potentiometer. In this setup, I2S and analogread() drivers seem to get into conflict, and they stop working.

The best solution is actually to use I2S for sound input, for example attach IMNP441 which is well-rested and provides far superior quality than any directly attached analog signal.

One idea - however just a wild guess - would be to set "disable wifi sleep" in wifi settings. It might make a difference.

softhack007, 2022-11-01:
> This is solved by setting a small guard interval before resetting the I2S peripheral in usermod.cpp:
> ```
>  // Reset I2S peripheral for good measure
>   i2s_driver_uninstall(I2S_NUM_0);
>   delay(100); // Give that peripheral some time to disable.
>   periph_module_reset(PERIPH_I2S0_MODULE);
> ```

> Frank, if you have the opportunity to test this and make changes to the code, I would be very happy about it.

The code you suggest ist already there since some time. I could add a short delay, but I don't belief it will change much. Espressif have confirmed that only a "hardware reset (power-on)" can disconnect the ADC from the I2S unit.

https://github.com/atuline/WLED/blob/086b5165bf90cc17f71c9b9f9604ffb0df6456da/wled00/usermod.cpp#L27-L31

softhack007, 2022-11-01:
Edit: Does the extra `delay(100);` solve it for you? If yes, please make a PR and i'll merge it into `dev`.

phonec, 2022-11-01:
that's right, the delay that already exists after periph_module_reset() is also necessary for proper initialization.

I'm just doing an additional guard delay of 100ms between i2s_driver_uninstall and periph_module_reset, as this solves the problem of falling off the analog microphone during a soft reset and changing the pin as I wrote, please check, for me it solved the problem, I tested the soft reset all day and every time the microphone remained active.

softhack007, 2022-11-01:
OK, that's good enough. Actually, timing is not critical in that part of the code.
Please make a PR and i'll merge it.

softhack007, 2022-12-19:
In case that analogread() seems to work but WLED animations still don't react, the reson might be that the analog input signal is not centered around 1.8V (analogread would return 0 in silence, while it should be around d 2048 if the signal is well centered).
See discussions here https://github.com/MoonModules/WLED/issues/10

jonandel, 2022-12-29:
I recently upgraded from v 12.0 to current latest 13.3 - and I now see an similar problem on (cold) boot about 60% of the time - sound reactive features simply dont work: no resonse to analogue mic stimulation.  I may be dreaming, but stimulating the mic during the boot cycle appears to make it 'stick' working, but if I dont create noise during the boot process (within 1 seconds or so) it appears to simply fade out, or set the gain to 0....  will test more and provide details.

softhack007, 2023-06-25:
Situation might improve with the last beta 0.13.4

* https://github.com/atuline/WLED/releases/tag/v0.13.4-beta

If still not working, then -sorry- you might have a broken max4466. It works for many other users, including myself.

Any way the MAX4466 is very bad quality, if you need to use analog go for max9814. Even better try out an I2S microphone like the INMP441.

Closing as "cannot reproduce"

closed 2023-06-25 by softhack007