atuline/WLED #233: 24 Key & 44 Key IR remotes do not work

cr4nkz, 2022-09-27:
### What happened?

Hello everyone. I tested the IR capability of WLED sound reactive 0.13.2 against WLED 0.13.3. Original WLED does work flawlessly with IR Remote (24 and 44 Key), but NOT with Sound Reactive WLED. I specified IR Pin and remote type on both tries. Platform is an ESP32 Is that a bug?

### To Reproduce Bug

Test WLED sound reactive 0.13.2 against WLED 0.13.3 with IR remotes on ESP32.

### Expected Behavior

Hex-Codes get recognized, but LEDs are not reacting.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-11-07:
Please try with our current [master branch](https://github.com/atuline/WLED/tree/master) (SR 0.13.3).

You need to modify wled00/wled.h to re-enable IR Remote support, then build the firmware yourself in platformIO.

https://github.com/atuline/WLED/wiki/Installing-and-Compiling#compiling-from-platform-io

IR remote has been disabled explicitly in SR WLED,  due to negative performance impact on soundreactive functionality.

softhack007, 2022-11-12:
The IR code in SR WLED is exactly the same as in upstream WLED 0.13.3. So the mist likely reason is that our standard firmware build is not including IR support.
A warning message has been added in the LED config page with commit https://github.com/atuline/WLED/commit/3f63e5424572822bd5dd7ae793df9209ca692673.

@cr4nkz in case you think the problem still exists, please respond within the next week.

softhack007, 2022-11-20:
Loses due to inactivity.

closed 2022-11-20 by softhack007