atuline/WLED #235: WIKI not included in web browser search

netmindz, 2022-10-09:
Github actively block indexing of any of the wiki, which makes finding information harder than it should be (https://github.com/community/community/discussions/4992)

Solution - Run export to static HTML and host somewhere - maybe github pages?

https://github.com/yakivmospan/github-wikito-converter

See attached example of output with default settings 
[example.zip](https://github.com/atuline/WLED/files/9741330/example.zip)

netmindz, 2022-10-09:
PR created https://github.com/atuline/WLED/pull/236

See example from my own project https://netmindz.github.io/balboa_GL_ML_spa_control/

Not the pretty, but at least will be indexed by Google

netmindz, 2022-11-27:
Nudge @atuline

ewoudwijma, 2022-11-27:
What do you think about this?
A fork of kno.wled.ge so perfect symbiose of upstream info + fork info IMO.

Supported by GitHub workflows so any commit towards it will automatically update the pages.

atuline, 2022-11-27:
I"m up for all that!!

netmindz, 2022-11-28:
@ewoudwijma looks nicer and should be indexable by Google, though that's for the newer MoonModules rather than the 0.13 maintenance release here.

Should I raise an issue against MM to copy over the info that's on this wiki that's missing there, e.g the page about the digital Mic hookup?

ewoudwijma, 2022-11-28:
> Should I raise an issue against MM to copy over the info that's on this wiki that's missing there, e.g the page about the digital Mic hookup?

At the moment https://moonmodules.github.io/WLED-Docs/ is in first setup stage, we still need to think about the best setup. I did some changes today and tried to capture that here: https://github.com/MoonModules/WLED/issues/6

@netmindz , Can you take a look, and add your thoughts?