atuline/WLED #237: Relay pin not working

cr4nkz, 2022-10-13:
### What happened?

0.13.2 Firmware does not react to Relay Pin.

### To Reproduce Bug

Flash firmware and measure voltage between GND and specified Relay pin. Its a few mVs but not 3.3V

### Expected Behavior

When Relay Pin is specified as non inverting, pin should be pulled high to 3.3V when ON. OFF should be 0V, vice versa with inverted checkbox checked.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.2

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2022-10-13:
Have you had this hardware setup working on previous WLED version. If so please state which

softhack007, 2022-11-12:
@cr4nkz SR relay code is exactly the same as in upstream WLED 0.13.3. Please check if you see expected behaviour with [upstream WLED 0.13.3](https://github.com/Aircoookie/WLED/releases/tag/v0.13.3).

In case that upstream WLED also shows the same unexpected behaviour, please open an [issue report for Aircoookie WLED](https://github.com/Aircoookie/WLED/issues/new/choose), to ensure the problem will be addressed at the right level.

linkua, 2022-11-24:
Try adding a pull-up resistance to the pin

closed 2022-11-28 by softhack007