atuline/WLED #240: Is it possible to set matrix width and height only for specific segment

lokster, 2022-10-22:
I have a setup where I drive multiple strips from one ESP32. One of the outputs drives a matrix, the others are strips with different length.
Is it possible to configure the 2D matrix width and height only for specific segment?

softhack007, 2022-10-22:
No.

lokster, 2022-10-23:
Is it because of architectural reasons, or it just needs a little bit of coding?
Since each segment can have different effect running on it, it makes sense (at least to me) to be able to configure only a certain segment as a matrix, so the 2D effects run properly on it.
My particular use case, is a custom ESP32 controller for my gaming PC case, for which I used the upstream WLED until now. I control several strips with it (including on rgb fans), as well as a big 16x8 matrix on the front panel. It works very nice, and I can even control only the matrix via Hyperion from the pc. But I like being able to use it standalone, without any pc control.
So it would have been great to be able to make the 2D effects work properly on the segment which controls the matrix.
I saw that the "Segment" struct in the code has "width" and "height" properties which are used when the effect runs. I tried some crude hard-coded patching - just to see if it can be done (even if not perfect), but so far without much success :)

kllaus333, 2022-10-29:
Hi lokster , i was trying something also without succes , i am looking also for the solution , please if you find something write back here :)