atuline/WLED #241: Sound Reactive issues with firmware update - forcing to update to NO sound reactive

homonto, 2022-10-25:
### What happened?

installed SR one ( WLED 0.13.0-b6 (build 2112080) and in Home Assistant "firmware update" message happens. When you click to update, it updates to NO sound reactive firmware.

### To Reproduce Bug

click on "firmware update available" in home assistant

### Expected Behavior

if there is new SR version - update to new firmware version BUT SR
if not - don't show the message

### Install Method

Binary from WLED.me

### What version of WLED?

 WLED 0.13.0-b6 (build 2112080)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-10-25:
Your problem report seems to be related to a very old release of SR WLED. 
The problem you describe might be related to the default firmware update link included in our software. This link was corrected in SR WLED 0.13.2.

Our latest release is available for download here: https://github.com/atuline/WLED/releases
SR WLED releases are also included in this web-based installer: https://wled-install.github.io/

Home Assistent integration is a feature of standard WLED, and not specific to our soundreactive functionality.

Closing, as the problem is already resolved in our current release.

Please open a problem report for "standard WLED", if you continue to experience problems: https://github.com/Aircoookie/WLED/issues/new/choose

<br/>

Please consider joining the WLED Discord groups where we have dedicated channels to discuss this project, your projects, 
and answer any questions you may have.

[image]
Join Discord to discuss beta testing of our [sound reactive fork](https://discord.gg/4CQRmfR) of WLED. 

[image]
Join Discord to discuss [AirCookie's WLED](https://discord.gg/KuqP7NE).

closed 2022-10-25 by softhack007

homonto, 2022-10-25:
yeap, I just upgraded to 0.13.2 and I don't see this problem
however, after upgrade WLED is not anymore visible in HA - I can only control it over the direct link to IP of WLED

homonto, 2022-10-25:
thank you, are you saying then that since I upgraded to 0.13.2 I will not be able to control WLED from HA? because the old version (0.13.0-b6) was visible in HA

softhack007, 2022-10-25:
sorry, I don't know. Please ask in discord, maybe another user knows the answer.
Some people have reported that deleting and re-adding in HA helps, so you could try that first.

homonto, 2022-10-25:
thank you, that was first what I tried but no luck
I will try over MQTT then ;-)
great job WLED is but without possibility to control from HA is kind of not working for me
I will try more then

softhack007, 2022-10-25:
BTW, you need a specific build with MQTT enabled for that. 

Standard SR is not including some extended functionality like MQTT, IR, Alexa, Bynk, and a few more extras . Reason is performance, because we need all available processing power for audio analysis. 

See our Wiki https://github.com/atuline/WLED/wiki
> We have also disabled functionality for other interfaces, such as Alexa, Blynk, Cronixie, Huesync, Infrared. If you would like to enable them, you will need to modify wled.h, compile and upload to your device.

homonto, 2022-10-25:
ha, I just found it somewhere that I need another build
do I need to compile myself or can I find somewhere there? ;)

softhack007, 2022-10-25:
you might find something in one of these locations:

* https://wled-install.github.io/
* https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/Sound_reactive
* https://github.com/wled-install/wled-install.github.io