atuline/WLED #245: Cannot change the LED Preferences

Patrickpopp, 2022-10-31:
### What happened?

I connnot change the led preferecnes. When i make changes and hit save nothing happens and then when i go back and back in the changes are gone. In the main branch it works so now if i whant to change something i flash the main branch change the settings and change back.

### To Reproduce Bug

I use the quinled dig quad and after configuring 3 outputs with about 400 leds each i am not able to change the led preferences anymore

### Expected Behavior

I can make changes on the led preferences and save them.

### Install Method

From SoundReactive Release Page https://github.com/atuline/WLED/releases

### What version of Soundreactive WLED?

WLED SR version 0.13.3 (build 2210301)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-10-31:
Could it be that you tried to use a GPIO pin that is already configured in sound setting or other usermod settings? 
Like GPIO 14, 15, 32 or 36?

Patrickpopp, 2022-10-31:
> Could it be that you tried to use a GPIO pin that is already configured in sound setting or other usermod settings? Like GPIO 14, 15, 32 or 36?

I´m using pins 16, 3 and 1

softhack007, 2022-10-31:
Sorry I still cannot reproduce  your problem. I've just configured 4 LED strips on my generic ESP32 board, then changed segments, then changed LED setting. No problem. In general, there is not difference between "upstream" WLED 0.13.3 and SR 0.13.3 in the LED preferences.

Can you give some more details?
* Screenshot of your LED preferences page
* Screenshot of SR WLED Info page
* What GPIO did you try to add (or change) as LED PIN?
* Anything else that might be special about your setup? 
* Did you install clean via USB (including erase_flash), or did you update via OTA?

Patrickpopp, 2022-10-31:
I installed it via OTA.
I cannot save it when i do any change like boot preset, led length or diffrent pins.
[image]
[image]
[image]
[image]

softhack007, 2022-10-31:
Please add the info page from [SR 0.13.3](https://github.com/atuline/WLED/releases/tag/v0.13.3)? 

It contains more information that could help us to understand what's going wrong.

Patrickpopp, 2022-10-31:
[image]

softhack007, 2022-10-31:
> i had one working and after loading that config in another one
Not sure if "loading the config into another controller" is supposed to work at all. Please check the WLED knowledge base.
Can you take one of your 23 controller, and start completely fresh? 
1. do a clean install of SR 0.13.3 from here: https://wled-install.github.io/ 
2. make all necessary setting in the Web UI - without restoring anything

Patrickpopp, 2022-10-31:
I used the Backup configuration an load configuration Feature. I am not on sight so I cannot reflash them at the moment. But I try to reproduce the error on one esp I have here

softhack007, 2022-10-31:
Edit: it might help if you could upload your `cfg.json` and `presets.json` from the controller that shows the problem. No worries about wifi password - they are stored in another file that is not included when you make a backup.

<br/>

you might also want to join the discord, to get help from the WLED community. Maybe there is someone who already experienced your problem and knowns how to best solve it.

[image]
Join Discord to discuss beta testing of our [sound reactive fork](https://discord.gg/4CQRmfR) of WLED. 

[image]
Join Discord to discuss [AirCookie's WLED](https://discord.gg/KuqP7NE).

Patrickpopp, 2022-10-31:
[presets.txt](https://github.com/atuline/WLED/files/9901234/presets.txt)
[cfg.txt](https://github.com/atuline/WLED/files/9901235/cfg.txt)

softhack007, 2022-10-31:
OK, thanks. One idea: as GPIO#1 is also "Serial TX" and we have very few Serial.print() still in the SR code and in external libraries ... 

If you avoid using GPIO#1 for LEDs, does that make a difference?

Patrickpopp, 2022-10-31:
when i use 16, 3 and 4 it also breaks. but 3 ist also a serial pin if i read the pinout correctly

Patrickpopp, 2022-10-31:
> when i use 16, 3 and 4 it also breaks. but 3 ist also a serial pin if i read the pinout correctly

but it also braks with 16,12 and 4

softhack007, 2022-10-31:
> but it also breaks with 16,12 and 4

so no luck ... guess we'll need to take some time and look in detail at your cfg.json and presets.json.

Patrickpopp, 2022-10-31:
okay, but it also happens with no presets so it schoul only be the config. here it is from my last try
[cfg (3).txt](https://github.com/atuline/WLED/files/9902418/cfg.3.txt)

softhack007, 2023-01-06:
Update: I can reproduce the behaviour.
* take a device running WLEDSR v0.13.3 dev
* upload presets.json from https://github.com/atuline/WLED/issues/245#issuecomment-1297056595
* upload cfg.json from here https://github.com/atuline/WLED/issues/245#issuecomment-1297248168
* reboot, then connect to WLED-AP
* go into LED preferences
* click on "save" -> no reaction

The following error appears in IE debug console when I click on "save" in LEDs settings:
[image]

https://github.com/atuline/WLED/blob/e7d5a04aa455410ae4519499abd01d29548942cd/wled00/data/settings_leds.htm#L548-L557

https://github.com/atuline/WLED/blob/e7d5a04aa455410ae4519499abd01d29548942cd/wled00/xml.cpp#L501-L504

@ewoudwijma can you take a look?

softhack007, 2023-01-06:
Update: 
After changing to ` max="15000"` in this line - and npm run build - It was possible for me to save the LED settings page.
https://github.com/atuline/WLED/blob/e7d5a04aa455410ae4519499abd01d29548942cd/wled00/data/settings_leds.htm#L550

However this will only cure the symptom, as max=1500 was surely picked for a good reason ....

softhack007, 2023-06-25:
Fix included in latest v0.13.4-beta release
* https://github.com/atuline/WLED/releases/tag/v0.13.4-beta

closed 2023-06-25 by softhack007