atuline/WLED #249: Peek colors are incorrect

T00LF00L, 2022-11-03:
### What happened?

Installed latest sound reactive version from the web installer and the colors shown when the peek tan is active are incorrect.

### To Reproduce Bug

Activate peek tab

### Expected Behavior

Colors should match LEDs.

### Install Method

From https://wled-install.github.io/

### What version of Soundreactive WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-11-03:
Can you give us some more details?
* what does "incorrect" mean for you? Too dim, too bright, red instead of green?
* what kind of LED stripe do you use
* what color order did you choose in LED settings (RGB, BGR, ...) ?

T00LF00L, 2022-11-03:
Right here you can see I make the lights blue, peek shows them green.  Everything was setup correctly, BGR with 2812 LEDs, I updated to latest release and not only is peek wrong, but many effects will not use random color cycle anymore, they’re all stuck on pink/blue shades only.
[image]

T00LF00L, 2022-11-03:
Here is fire effect, LEDs are correct fire color, peek is purple/blue.
[image]

T00LF00L, 2022-11-03:
So for example, when I use the DJ Lights effect with the mic, I’m only getting pink & white colors and some blue, nothing else.  Or some other effects, if I select random color cycle, I can see the colors cycling in the menu, but the LEDs stay on either shades of pink, or a green/yellow mix.  The color pallets seem to be broken and they don’t match what peek is showing either.

wojteksolo, 2022-11-09:
I have a similar issue after updating to version 0.13.3. If I choose RGB color order, the LED color follows the color wheel. If I choose GBR order, LED color matches Peek.
[image]

ewoudwijma, 2022-11-29:
Fixed in https://github.com/atuline/WLED/commit/ecb0d48a61f4ad719b0bb1c67bb82f5a786ef1e4

Issue was not Rgb/grb but peek1D not using the right led values.

@wojteksolo , @T00LF00L , can you let me know if it works now then I can close the issue

T00LF00L, 2022-12-07:
Yes sir, I confirmed this is fixed.  
Sorry I was so vague, this is extremely new to me.  The last time I looked at code it was on a Commodore 64..

ewoudwijma, 2022-12-07:
No worries, didn't seem vague to me ;-)
Code hasn't changed much since then ;-)

Closing the issue now