atuline/WLED #251: Random color cycle isn’t working

T00LF00L, 2022-11-11:
### What happened?

Random color cycle is only displaying a rainbow.

### To Reproduce Bug

Select random color cycle with any effect.

### Expected Behavior

Color should be a random color cycle.

### Install Method

Binary from WLED.me

### What version of Soundreactive WLED?

V0.13.3 “Toki+SR”

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-11-30:
Hi,

I can't reproduce the problem - maybe it is something specific to your setup, or you have different expectations?

[image]

What I did:
* select an effect that supports Palettes, like the "palette" effect
* click on "random cycle" (see screenshot)
* Wait ~1 minute, and I see that colors on my light changed several times.

Actually there is a note in the source code that random palettes will (sometimes) not work with multiple segments, due to a limitation of the library we use. 
Also some effects have a fixed color palette, or they use only the primary/secondary color but no palette at all.

https://github.com/atuline/WLED/blob/cdeecce2e0ccca92424b772d3c6bb354ee632aeb/wled00/FX_fcn.cpp#L1248-L1293

softhack007, 2022-12-24:
Closed, as I cannot reproduce and no additional information provided by the problem report author.

closed 2022-12-24 by softhack007