atuline/WLED #252: Apple/Google CarPlay Support

jterrell345, 2022-11-19:
**Is your feature request related to a problem? Please describe.**
I have used WLED for ambient lighting in my vehicle and it would be nice to configure it using Apple or Google CarPlay.

**Describe the solution you'd like**
When in carplay, I want to enable presets, change color and potentially use the microphone to pick up the beat of music in the car.

**Describe alternatives you've considered**
Giving my passengers access to the controller and letting them make adjustments on their own phone.

**Additional context**
I think a simplified UI for CarPlay would make this project more valuable because there is very little in the industry that has the capabilities of WLED.

Thank you for your ideas for making WLED better!

softhack007, 2022-11-19:
Hi, thanks for this interesting idea.
The functionality you describe could be added in "upstream" WLED, as it is not specific to soundreactive functionality.
Once it would be implemented in upstream, the feature comes to WLEDSR automatically when we merge in the latest upstream code.

Please create a feature request here: https://github.com/Aircoookie/WLED/issues/new/choose

Thanks.

softhack007, 2022-11-19:
Closed, as not planned for SR WLED. Please feel free to make a feature request for upstream "Aircoookie" WLED.

closed 2022-11-19 by softhack007