atuline/WLED #253: I2S not working

coolinger, 2022-11-20:
### What happened?

With commit f07b2aeba1f649c00257b3087bf4585b9fcf2f10, I2S is not working anymore.

### To Reproduce Bug

Checkout f07b2aeba1f649c00257b3087bf4585b9fcf2f10 and compile

### Expected Behavior

I2S working like before

### Install Method

Self-Compiled or other

### What version of Soundreactive WLED?

SR WLED Build 2211201

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
AS: Generic I2S Microphone - left channel only.
AS: Failed to initialize sound input driver. Please check input PIN settings.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-11-20:
Hi, thanks for the fast feedback :-)

I've quickly tested it before making that commit, maybe the new sanity checks are still too strict.

Can you please
* add a screenshot of your sound setting?
* recompile with `-D SR_DEBUG -D NO_MIC_LOGGER` added to your `build_flags` ? This will enable a few more debug messages that should help me to find out why the driver refuses to initialize.

softhack007, 2022-11-20:
Update: found it. Missing brackets around one of the new checks. Will commit a fix later today.
Thanks again for reporting this!

coolinger, 2022-11-20:
Thanks for quick reply. I really tried to find the bug myself but gave up eventually. After you wrote "missing brackets" i found it too (line 176?). But my git{,hub}-fu is too low to be of any further help ;)

Keep up the great work!

softhack007, 2022-11-20:
Yes, line 179, so the return statement was always executed even when no error happened.

Please check out commit https://github.com/atuline/WLED/commit/f8898367bdac4ce63a364868ac95286765e9d265 and tell me if it works for you again.

coolinger, 2022-11-20:
Works again, thanks!

closed 2022-11-20 by coolinger