atuline/WLED #256: Loxone Commands (LX) not working

Clubsport, 2022-12-02:
### What happened?

Hey guys,

it seems that the Loxone-specific commands (&LX=) are not working.
Installed the "normal" 0.13.3 from @Aircoookie and it works flawless.
Changed back to SR-013.3 and not reaction at all.

Can you please have a look on this? :)

### To Reproduce Bug

Changed between 0.13.3 and SR-0.13.3

### Expected Behavior

http-commands with direct input from Loxone home automation.

### Install Method

From SoundReactive Release Page https://github.com/atuline/WLED/releases

### What version of Soundreactive WLED?

SR WLED 0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-12-02:
Hi,
Loxone is not included in our standard SR build (long story - has to do with memory and cpu ressources...).

https://github.com/atuline/WLED/blob/cdeecce2e0ccca92424b772d3c6bb354ee632aeb/platformio.ini#L467

To get Loxone support in SR WLED, you need to build your own SR firmware - 
see  https://github.com/atuline/WLED/wiki/Installing-and-Compiling#getting-started .

<br/>

Unfortunately, we missed to add a message in Sync Settings page, like it was done for other disabled functions. The next SR release will show "This firmware build does not include Loxone Lighting support." when Loxone was not enabled at compile time.

Clubsport, 2022-12-02:
Ah okay, didn't found any hint on that. ;)
So I have to try to make my own one.... I will give it a try! :)

softhack007, 2022-12-16:
seems this is a problem in upstream WLED
see https://github.com/Aircoookie/WLED/issues/2957