atuline/WLED #259: FFT data for sound-reactive custom effects

protyposis, 2022-12-15:
I would like to be able to write sound-reactive custom effects based on frequency magnitudes. It would be great if the FFT data, e.g., `fftResult`, was made available to the effect scripts.

netmindz, 2022-12-15:
That's already there, see GEQ for an example

protyposis, 2022-12-15:
What is "GEQ"? I checked the [wiki](https://github.com/atuline/WLED/wiki/WLED-Custom-effects), [custom effects repo](https://github.com/MoonModules/WLED-Effects/tree/master/CustomEffects/wled), and [script interpreter source](https://github.com/atuline/WLED/blob/744dc734f4a70f479b7f93f1757dd4086392cf41/wled00/src/dependencies/arti/arti_wled.h), but wasn't able to identify this functionality.

softhack007, 2022-12-15:
GEQ = graphical equalizer - one of the 2D effects in fx.cpp.
I think that @protyposis means to say that the same data (fftResult[]) cannot be used in the custom effect scripting language.

protyposis, 2022-12-15:
Exactly! I updated my earlier comments for clarity.

ewoudwijma, 2022-12-16:
It's done, see here: https://discord.com/channels/700041398778331156/905206400349765642/1053293141286998080

🥳

protyposis, 2022-12-16:
I'm not registered there but have seen the video - this is great, thank you! 🚀 Can't wait to use it :)

ewoudwijma, 2022-12-16:
It's now in WLED SR 0.13.3.6 and WLED MM 0.14.0.11.

If you download the new wledv033.json and presets.json and fftBrightness.wled after you press the Custom Effects button in Segments you can reproduce the effect.

Please try it out and if you create a new fft based effect, please share so we can add that to the library

softhack007, 2023-06-25:
Included in latest v0.13.4-beta release
* https://github.com/atuline/WLED/releases/tag/v0.13.4-beta

closed 2023-06-25 by softhack007