atuline/WLED #260: Analog Microphone (MAX9814) Not Working on 0.13.3

ali-essam, 2022-12-22:
### What happened?

Sound reactive effects not working on ESP32 with Analog Mic MAX9814 (connected the same way as in the docs).

I first flashed 0.13.3 via esptools and everything works except the sound reactive effects, tested on different ADC1 inputs, but it won't work.

There is a weird behavior though that right after a powercycle or a reset, when switching the effect to a SR effect the first time, it'll pick up a faint signal that will go down in 2 or 3 seconds, for example with noise meter, it'll show up 7 leds (/144) that will go down to zero leds in couple of seconds. After that, nothing else is picked up until it's reset/cycled.

Reverted back to 0.13.0-b4 as mentioned in #162 through OTA and it was working fine, upgraded again to 0.13.3 over OTA and it stopped working again.

I don't have any buttons or any other input, just the esp32, the microphone, powersupply and the leds

The board in use: AZDelivery ESP32 NodeMCU Module WLAN WiFi Development Board with CP2102

### To Reproduce Bug

- Connect analog microphone (max9814) to esp32
- Flash WLED SR 0.13.13
- Select an SR Effect

### Expected Behavior

SR effects work the same as they're working in previous releases

### Install Method

From SoundReactive Release Page https://github.com/atuline/WLED/releases

### What version of Soundreactive WLED?

WLED 0.13.3 (build 2210301)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2022-12-22:
Just to confirm, are you saying audio responsive patterns work correctly with your current hardware setup if you run 0.13.0-b4?

ali-essam, 2022-12-22:
Yes, the same exact setup, no changes, works perfectly on `0.13.0-b4` but not on `0.13.3`

ali-essam, 2022-12-22:
I have also now tried `0.13.0-b6` and `0.13.2` but not working too, so the issue is definitely something the changed between `0.13.0-b4` and `0.13.0-b6`

ali-essam, 2022-12-23:
@netmindz Is there a way to debug this on my side?

gonzokalle, 2022-12-24:
I have the same problem with 0.13.0-b6 and higher. The analog Microphone working only a short time.
With 0.13.0-b4 works good.

softhack007, 2022-12-24:
See #231

iamSIAM, 2022-12-25:
I have sae issues ,Thats why  i dont updated & using older version of sr wled!!!  I was thinking this is microfon compabilty problem in higher version ..

softhack007, 2022-12-25:
It is not a generic compatibility issue, but something special that seems to happens to some users. There are a lot of poeple (including myself) who use max4466 or max9814 with the latest release, and it works for them without problem. OK, poor quality compared to I2S,  but the mic works as expected.

So it could be related to other factors, like 
* esp32 hardware (adc, power supply) 
* bad wiring, or voltage fluctuations
* if your mic delivers "0" when you perform analogread() in silence, then you have a  broken microphone as input should be centered around 1.8V and not 0V
* some microphone  batches are just no good

Actually there will be an update of SR end of January with a new filtering method for analog mics. There is some chanche that this new filter will work better with bad analog mics that are not centered peoperly. So - maybe - the upcoming release will help you. The new filter is already activated in our `dev` branch, so you could try a nightly build from the development branch, and see if that helps.

ali-essam, 2022-12-27:
I'll give it a try later 👌 

As a feature request, is it possible to integrate those readings for debugging in the microphone settings view in the future? so it would be easier to check what's wrong with the microphone without having to tinker with Arduino

jonandel, 2022-12-31:
(moved from 13.2 issue thread).
I recently upgraded from v 12.0 to current latest 13.3 - and I now see an similar problem on (cold) boot about 60% of the time - ie sound reactive features simply dont work: no response to analogue mic stimulation. I may be dreaming, but stimulating the mic during the boot cycle appears to make it 'stick' working, but if I dont create noise during the boot process (within 1 seconds or so) it appears to simply fade out, or set the gain to 0....
Board is an ESP32-WROOM v1; Gneric Analogue mic.  
I can measure vcc and the output to the analogue pin - looks fine (by multimeter, not scope).  But WLED simply reports as 'mic - quiet'.
I spent about a day looking into PSU, wiring, etc, different ESP32's - but in the end gave up (sorry!), as I had to use the device. 
I ended up installing a digitial mic, which works flawlessly, and is more responsive anyway.

softhack007, 2023-06-25:
Situation might improve with the last beta 0.13.4

* https://github.com/atuline/WLED/releases/tag/v0.13.4-beta

If still not working, then -sorry- you might have a broken max9814. Its works for many other users.

Closing as "cannot reproduce"

closed 2023-06-25 by softhack007