atuline/WLED #261: What will happen to this branch?

TysTies, 2022-12-23:
### What happened?

I've used your branch for years and find it more reliable than the original WLED. If they are implementing sound reactive, what will happen to your branch? https://github.com/Aircoookie/WLED/releases/tag/v0.14.0-b1

### To Reproduce Bug

n/a

### Expected Behavior

 n/a

### Install Method

From SoundReactive Release Page https://github.com/atuline/WLED/releases

### What version of Soundreactive WLED?

WLED 0.13.3 (build2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
n/a
```

### Anything else?

 n/a

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ewoudwijma, 2022-12-23:
Exactly for that reason we will keep the branch alive, synced with laters 0.13.x from upstream but not with 0.14. We created a separate 0.14 branch called MoonModules where the sr team will continue to do things which cannot be done in upstream:

WLED SR: stable / maintenance 0.13.x based
WLED AC: will (soonish?) move from 0.13.x to 0.14
WLED MM: Bleeding edge new experimental stuff which might or might not be merged to upstream

softhack007, 2022-12-24:
See https://github.com/atuline/WLED/issues/231

TysTies, 2022-12-24:
Awesome! I love what you guys are doing! I look forward to your 3D Cube experiments as well!

TysTies, 2023-01-20:
When can I expect SR WLED 14.0? I like the settings lock feature, in the new version of the main branch. Now I can display my creations at the fair and let people use their mobil device to control them, rather than supply a tablet.

softhack007, 2023-01-20:
Actually SR WLED 0.14.0 = upstream WLED + audioreactive usermod; 

so you could try 
* either the MoonModules fork (latest development based on 0.14.0, including audioreactive and custom effects), 
* or plain "upstream WLED" with audioreactive usermod.

https://wled-install.github.io/

softhack007, 2023-01-20:
We are also planning a maintenance release of SR 0.13.3, with a number of bugfixes and stability improvements, in a month or so.

TysTies, 2023-01-23:
Where do I find the user mod?