atuline/WLED #264: PCM1802 FSYNC

pr8x, 2023-01-15:
I am trying to make the PCM1802 work with WLED-SR, but I am not sure what to do with the FSYNC pin. According to the datasheet:

> FSYNC enables the BCK signal, and the device can shift out the converted data while FSYNC is HIGH

Does the "Generic I2S with MCLK" driver support this somehow? If not, it seems to quite easy to implement.

softhack007, 2023-01-15:
Actually the espressif I2S driver that we use has no support an an additional "fsync" pin, so no idea how that could be supported.

A user on discord has reported that he connected a PCM1808 with success, maybe you find some helpful information there: https://discord.com/channels/700041398778331156/853969074311659570/1037034721101750334

<br/>

Join our Discord to discuss sound reactive WLED, and get help and support from the community:

[image]

softhack007, 2023-04-09:
Hi again, I just stumbled over this website: https://www.pjrc.com/pcm1802-breakout-board-needs-hack/

It seems that "FSYNC" is only relevant in TDM mode (an extension of I2S for multiple microphones). So maybe you just need to connect FSY to 3.3V.

> The FSY pin (connected to FSYNC) is also a bit unusual.  PCM1802 expects it to be logic high while you transmit data, so just connect to 3.3V.  In the other modes, it sends a signal on this pin which is high during data bits and low during the zero padding bits.  But it does not require that signal as input.  FSYNC just connects to 3.3V to use PCM1802 with Teensy.

pr8x, 2023-05-01:
@softhack007 I ended up just using the PCM1808.