atuline/WLED #267: Boot loop AZ-Delivery ESP32-Wroom-32

pmalsenz, 2023-05-30:
### What happened?

Hi,
after successfully compiling the firmware for the task soundReactive_esp32dev from the atuline WLED master branch and flashing it to the MCU, it immediatly enters a boot loop (nearly like #211, but without the Brown out message). Same happens with the latest binary release 0.13.3.

With the latest Aircoookie WLED source compiled from the main branch, the controller works without flaws.
WiFi connects, the UI is reachable and I can set my preferred colors etc.
There's no difference if the MCU is powered via USB (5V+ disconnected, only GND connected) or if powered by the Meanwell power supply without connection to my computer.

### To Reproduce Bug

At the moment my controller is powered via USB from my build computer and the 5m BTF-Lightning WS2812b LED Strip with 300 LEDs is powered by a Meanwell RS-150-5 power supply (5V/150W). After flashing and rebooting the MCU, nothing else happens.
Same behaviour if the MCU is powered by the Meanwell power supply.

MCU: ESP32-Wroom-32 [(ESP32 NODEMCU Module WiFi Development Board with CP2102)](https://www.az-delivery.de/en/products/esp32-developmentboard)
Mic: INMP411 [(TECNOIOT 2 x INMP441)](https://www.amazon.de/dp/B07YXF6ZV2?language=en_GB&currency=EUR)
LED Strip: BTF Lightning WS2812b Strip 300 LEDS [(BTF-LIGHTING WS2812B LED Strip, 5 M Length, 60 LEDs/Pixels/m, 300 LEDs)](https://www.amazon.de/dp/B01CDTEKAG)

### Expected Behavior

MCU starts, connects to WiFi and WLED can be configured/used via Browser and/or Smartphone app.

### Install Method

Self-Compiled or other

### What version of Soundreactive WLED?

SR WLED 0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5388
entry 0x4008067c
AS: Analog Microphone (left channel only).
Ada
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

netmindz, 2023-05-30:
Can you run a build with WLED_DEBUG enabled

pmalsenz, 2023-05-31:
Sorry for my delayed reply, work got in the way of testing ;-)

I did the build with WLED_DEBUG enabled in wled.h after running the tasks _clean_ and _clean all_ for the soundReactive_esp32dev task.

Before I flashed the MCU, I did _erase_flash_ from the menu in vscode, flashed the WLED_DEBUG enabled firmware and I don't know how or why. It work's !? :-)

Here's the debug output from booting after reset:
```
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5388
entry 0x4008067c

---WLED 0.13.3 2210301 INIT---
esp32 v3.3.6-16-gcc5440f6a2
arduino-esp32 v1.0.x

CPU:   ESP32-D0WDQ6 rev.1, 2 core(s), 240MHz.
FLASH: 4MB, Mode 2 (DIO?), 80MHz

heap 257412
PIN ALLOC: Pin 1 allocated by 0x89 (137)
Registering usermods ...
Reading config
Reading settings from /wsec.json...
JSON buffer released. (3)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 4 allocated by 0x82 (130)
Successfully inited strip 0 (len 300) with type 22 and pins 4,255 (itype 17)
PIN ALLOC: Pin 0 allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
Initializing strip
Usermods setup
AS: Generic I2S Microphone - left channel only.
PIN ALLOC: Pin 15 allocated by 0x8d (141)
PIN ALLOC: Pin 32 allocated by 0x8d (141)
PIN ALLOC: Pin 14 allocated by 0x8d (141)
FFT running on core: 0
Access point disabled.
Connecting to HomeNET...
[AS userLoop] hickup detected -> was inactive for last 712 millis!

Connected! IP address: 192.168.x.y
Init STA interfaces
mDNS started
[AS userLoop] hickup detected -> was inactive for last 26 millis!
send NTP
NTP recv, l=48
Local time: 18:39

```

Here's the repeated debug messages from the vscode monitor:
```
---DEBUG INFO---
Runtime: 30000
Unix time: 1685551199,035
Free heap: 155316
Wifi state: 3
State time: 0
NTP last sync: 3593
Client IP: 192.168.x.y
Loops/sec: 1758
Max UM time[ms]: 1
Local time: 18:40
```

My first thought after successfully booting with debug enabled was, the firmware won't boot without connected microphone. But after reviewing the presets in the config menu, I saw that the mic preset is Generic Analog, so the MCU  could not know that there is aleady a digital mic connected

Atm I have no idea what caused the issue.

pmalsenz, 2023-05-31:
I just did a new build with disable WLED_DEBUG. Now it's the same as before, stops booting with the message above.

Is there anything I can post from the build process in vscode, that could help to identify the problem ?

```
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5388
entry 0x4008067c
AS: Generic I2S Microphone - left channel only.
Ada
```

softhack007, 2023-05-31:
Could you try our latest source code from the 'dev' branch?
It's very stable already, and it might work better.

https://github.com/atuline/WLED/tree/dev

Also try to build with '-D WLED_DISABLE_ADALIGHT', in case you have a "dangling" RX pin that is not pulled-down on the board.

pmalsenz, 2023-05-31:
Same behaviour as with the 'master' branch. The MCU boots then (the Superblock message disappears after a reset)
I haven't configured anything like WiFi, so a _pure_ build. Only with '-D WLED_DISABLE_ADALIGHT'
```
-- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1084
load:0x40078000,len:11220
load:0x40080400,len:5388
entry 0x4008067c
.pio/libdeps/soundReactive_esp32dev/LittleFS_esp32/src/lfs.c:1078:warn: Superblock not found (dir pair at {0x0, 0x1}). Filesystem needs formatting.
AS: Analog Microphone (left channel only).
Ada
```

pmalsenz, 2023-05-31:
Here's the output of a fresh build after running 'clean' and 'clean all' inside the soundReactive_esp32dev task and 'clean all' and 'clean all all' in the default task (just removed some path names and mac):

```
 *  Task wird im Ordner WLED_Atuline_Dev ausgeführt: platformio run --target upload --target monitor --environment soundReactive_esp32dev 

Processing soundReactive_esp32dev (board: esp32dev; platform: espressif32@3.5.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
*** use existing my_config.h ***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 0.0.0+sha.4b0c087 
 - tool-esptoolpy @ 1.30100.210531 (3.1.0) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
Converting wled00.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- FastLED @ 3.5.0
|-- IRremoteESP8266 @ 2.8.2
|-- ESPAsyncWebServerAircoookie @ 2.0.7+sha.b9349dc
|-- LittleFS_esp32 @ 1.0.6+sha.820c01e
|-- NeoPixelBus @ 2.6.9
|-- AsyncTCP @ 1.2.0+sha.38b8475
|-- arduinoFFT @ 1.9.2+sha.419d7b0
|-- Wire @ 1.0.1
|-- EEPROM @ 1.0.3
|-- ArduinoOTA @ 1.0
|-- DNSServer @ 1.1.0
|-- ESPmDNS @ 1.0
|-- WiFi @ 1.0
|-- ESP32 Async UDP @ 1.0.0
|-- WebServer @ 1.0
Building in release mode
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/FastLED.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/bitswap.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/colorpalettes.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/colorutils.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/hsv2rgb.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/lib8tion.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/noise.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/platforms.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/platforms/esp/32/clockless_rmt_esp32.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/power_mgt.cpp.o' from cache
Retrieved `.pio/build/soundReactive_esp32dev/libc24/FastLED/wiring.cpp.o' from cache
Compiling .pio/build/soundReactive_esp32dev/src/FX.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/FX_fcn.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/alexa.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/blynk.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/button.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/cfg.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/colors.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/dmx.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/e131.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/file.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/hue.cpp.o
Compiling .pio/build/soundReactive_esp32dev/src/improv.cpp.o
/.../PlatformIO/Projects/WLED_Atuline_Dev/wled00/cfg.cpp: In function 'bool deserializeConfigSec()':
/.../PlatformIO/Projects/WLED_Atuline_Dev/wled00/cfg.cpp:994:14: warning: variable 'interfaces' set but not used [-Wunused-but-set-variable]
   JsonObject interfaces = doc["if"];
              ^
/.../PlatformIO/Projects/WLED_Atuline_Dev/wled00/cfg.cpp: In function 'void serializeConfigSec()':
/.../PlatformIO/Projects/WLED_Atuline_Dev/wled00/cfg.cpp:1041:14: warning: variable 'interfaces' set but not used [-Wunused-but-set-variable]
   JsonObject interfaces = doc.createNestedObject("if");
[... 372 lines cut ...]
.pio/libdeps/soundReactive_esp32dev/LittleFS_esp32/src/lfs.c:1078:warn: Superblock not found (dir pair at {0x0, 0x1}). Filesystem needs formatting.
Preset file not found, attempting to load from EEPROM
JSON buffer released. (8)
deEEP complete!
Reading config
Reading settings from /wsec.json...
JSON buffer released. (3)
Restore settings from EEPROM
EEPROM settings invalid, using defaults...
Writing settings to /wsec.json...
JSON buffer released. (4)
Writing settings to /cfg.json...
JSON buffer released. (2)
Initializing strip
No busses, init default
PIN ALLOC: Pin 2 allocated by 0x82 (130)
Successfully inited strip 0 (len 30) with type 22 and pins 2,255 (itype 17)
Usermods setup
AS: Analog Microphone (left channel only).
PIN ALLOC: Pin 36 allocated by 0x8c (140)
FFT running on core: 0
No connection configured. Opening access point WLED-AP
Init AP interfaces
[AS userLoop] hickup detected -> was inactive for last 916 millis!
```

netmindz, 2023-05-31:
Ada is normal for final message, so unless it's looping, just try and access the device over WiFi

This caught me when in started that there isn't a final boot up all good message

netmindz, 2023-06-01:
When you say WiFi to connect, it has no credentials yet so can't connect to your network.  Do you mean that you don't see the WLED-AP appear on your phone/laptop/tablet

pmalsenz, 2023-06-01:
Sorry, yes that's what I meant. No WLED-AP.

netmindz, 2023-06-02:
Not sure why you only see the AP when debug enabled. No harm leaving debug on. 

This fork is for legacy support, you might be better trying MoonModules as you then get version 0.14

pmalsenz, 2023-06-02:
The MCU is actually running fine with debug enabled. As it's a personal fun project, I don't have a problem if I need to rebuild an reflash every now and then :-)

I'd say we close this issue and if I get any clue about it, I report it here.

Thanks for your support and your work !!!

closed 2023-06-02 by pmalsenz