atuline/WLED #272: gap file support

kenai2011, 2023-09-06:
usually need for special led-setups it would be great to have this feature.
Is it planned to implement?

Thanks
kenai2011

softhack007, 2023-09-06:
Hi, a similar feature is implemented in upstream WLED, called "gap maps".
This repo is in "maintenance mode" (bugfixing and stability improvements only), and we don't plan to add new features.

You could try the MoonModules fork which includes latest developments, based on latest WLED 0.14.0 source code.
* https://github.com/MoonModules/WLED/releases

closed 2023-09-06 by softhack007

kenai2011, 2023-09-06:
Hi Frank,

 

thank you, i will check it asap.

 

BR

Kenai

 
 

Gesendet: Mittwoch, 06. September 2023 um 13:24 Uhr

 

	https://github.com/MoonModules/WLED/releases