atuline/WLED #277: Current version no longer supported in Home Assistant Core update 2024.8.0

proosth, 2024-08-08:
After updating to Home Assistant Core 2024.8.0, all my sound reactive Athom WLED controllers are no longer compatible with the WLED integration. I also have multiple **non**-sound-reactive Athom WLED contollers that are able to update to the latest stable 0.14.4 "Hoshi" release. These are still working in the latest Home Asistant version.

Will this brance also be updated so we can all continue using it in the latest HA?

softhack007, 2024-08-08:
Hi @proosth,

This fork is not under development any more - the latest soundreactive WLED (based on 0.14.x) is in the MoonModules  fork now.

https://github.com/MoonModules/WLED

Actually I can't say much about the homeassistant integration - we (the MoonModules team) usually don't use HA. If 0.14.4 does work with HA, then chances are good that MoonModules WLED will work, too.

netmindz, 2024-08-08:
I have my WLED-MM instances registered with Home Assistant but don't actively use as all my setups are experimental rather than practical lighting for the home. We have far too much Hue for that already 🤣