atuline/WLED #278: Clarify the values encoded in SoundSync packets

jdtsmith, 2025-01-12:
**Is your feature request related to a problem? Please describe.**

To generate your own `SoundSync` packets separate from WLED, it would very valuable to know more about the units and values encoded in the packet, ideally in [this description of the format](https://github.com/atuline/WLED/wiki/UDP-Sound-Sync).  

**Describe the solution you'd like**

Add an entry-by-entry list which describes, for each member of the V2 `audioSyncPacket` struct:

1. What the value _means_ to WLED, i.e. how it is interpreted for sound reactive effects.
2. What units the value is measured in (if applicable).
3. What range of values it may possibly take.
4. What range of values it _typically_ takes, when produced by WLED.
5. Any other notes (whether it is optional, etc.). 

**Describe alternatives you've considered**

I've considered reverse engineering these values but would need debug console access to WLED UDP `SoundSync` received packets.

**Additional context**

One small confusion.  The Wiki cautions:

> this is a C language "struct". Due to padding performed by gcc, the actual package is slightly bigger

But it looks like this is an explicitly packed struct, with the gaps included verbatim.  Perhaps this note of caution is vestigial?

Thanks for your work on WLED!

netmindz, 2025-01-12:
This fork has been replaced by the MoonModules fork. 

You can find out more about the audio sync at https://mm.kno.wled.ge/soundreactive/sync/

softhack007, 2025-01-12:
@jdtsmith 

I've recently explained some more details of the format here: https://github.com/chrisgott/feed_my_wled/issues/1#issuecomment-2565423575

> "this is a C language "struct", but it looks like this is an explicitly packed struct

The statement is still valid, however we have recently  changed the definition for the V2 data structure to "packed struct", to make the gaps explicit. Byte offset in the structure have not changed.

jdtsmith, 2025-01-12:
Thanks.  Might be nice to add that info to the MoonModules page, along with some "observations" about the typical values and a bit more about what they constitute.

For example, you mention `value of biggest FFT result from a single FFT run`, but FFT codes differ in how they present "values".  Are "FFT values" the value in a specific channel in Volts (from a hypothetical mic), the _amplitude_ in Volts^2, or perhaps a decibel value, _relative_ to some fiducial amplitude?